mvp/uhubctl #258: jetson nano developer kit support

ethanrublee, 2020-07-31:
I've tested this on the jetson nano development board (B01 version).

My results:

bus 2-1 is the usb3.0 bus corresponding to the 4 ports on the dev kit.
[image]

- bus 2-1 port 1 can disable the VBUS of all the 2-1 ports.
- bus 2-1 ports 2-4 do not disable the VBUS.
- When plugging in another (unsupported) hub (anker https://www.amazon.com/gp/product/B07L32B9C2) to any port on bus 2-1, VBUS can no longer be disabled on bus 2-1.

mvp, 2020-07-31:
Thanks for reporting!

Regarding last case, you should be able to disable power anyway using `uhubctl -l 2-1 -p 1 -a 0 -r 100`. Try increasing -r option until it starts working.

ethanrublee, 2020-07-31:
Thanks @mvp, I tried with ``-r 1000`` and it has no effect.

mvp, 2020-07-31:
Thats weird, oh well... I guess you still have more or less working option of plugging external hub to ports 2 to 4.

closed 2020-08-01 by mvp

aslanpour, 2022-08-18:
It did not work for me. I had to do this instead
`echo '2-1' |sudo tee /sys/bus/usb/drivers/usb/unbind"`

mvp, 2025-07-01:
@grep-o, almost all embedded boards like Jetson or Raspberry Pi only support ganged power switching - that is, they have all USB ports connected to single power rail. If you turn off all ports feeding from the same power rail, it should turn it off (or remove VBUS).

If you require individual port control, you need to get external hub from uhubctl supported list that actually supports true per port power switching. Simply attach it to your Jetson board, and then you should be able to control all USB ports independently. One of good very small and cheap hubs that is friendly for embedded boards is UUGear Mega4. It works best for Raspberry Pi 4 and 5, but can be used with anything else like Jetson.

grep-o, 2025-07-02:
Thanks @mvp ,much appreciated!

commit c6927085 (2020-08-01): Add NVidia Jetson Nano B01 as supported device

Closes #258.