mvp/uhubctl #260: Allow uhubctl to run on incompatible hubs

wtolkien, 2020-08-05:
First, thanks for this very useful tool!

If I'm not mistaken, currently uhubctl looks at the 'Per-port power switching' flag to decide if a hub is compatible. 

I'd like to suggest that uhubctl no longer refuses to operate on incompatible hubs, but merely outputs a warning. 

The reason is that from my experience, even if a hub does not support real power switching, it still does a port reset which usually triggers a reset of the attached USB device. And that's a useful feature. 

This 'port reset only' actually happens a lot. Right now, very often a hub has the 'Per-port power switching' flag set without actually really supporting it. In this case uhubctl will already trigger a port reset. Not perfect, but the best it can do! 

Why not allow this on all hubs?

Thanks!

mvp, 2020-08-06:
Thanks for the idea, I will consider adding it. Not promising when though.

wtolkien, 2020-08-06:
Thanks - much appreciated! Alternatively there could be a '--force' option to run on incompatible hubs.

henzef, 2020-09-17:
I managed to switch a D-Link DUB-H4 F-Revision hub (which only supports ganged power switching) on and off with a hack. See https://github.com/mvp/uhubctl/issues/268#issuecomment-694095968

mvp, 2020-12-08:
Implemented in 0cc0929.

commit 0cc09299 (2020-12-08): Add a switch to allow operation on unsupported hubs

Show power switching type supported by a hub:

* ppps   - per-port power switching
* ganged - ganged power switching
* nops   - no power switching

Add --force (-f) switch to allow operation on non-ppps hubs.
Use `-f` at your own risk, this most likely will not work to actually switch power.

For ganged hubs, you may need to turn power off for all hub ports to get any effect.
For nops hubs, it is not likely to work at all, but might be useful for informational purposes.

Also, instead of printing recipe on how to fix USB permissions on Linux [... cut]
closed 2020-12-08 by mvp