mvp/uhubctl #261: AmazonBasics 7-Port Hub HU9003V1EBL ASIN:B076YKYHCB

c-MM, 2020-08-11:
I just got an AmazonBasics 7-Port Hub and wanted to let you know, that it works with uhubctl.
I need to specify -e and -l  and issue 2 commands to get correct port switching on the first hub 
otherwise it seems to pick the wrong pair.

Initial status:
```
root@z2srv:~# ~mm/uhubctl/uhubctl 
Current status for hub 2-1.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
Current status for hub 1-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
```
Switching port1 off:
```
root@z2srv:~# ~mm/uhubctl/uhubctl -l 2-1 -p 1 -a off
Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 00a0 off
Current status for hub 1-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
Sent power off request
New status for hub 1-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
```
As you can see it picks hubs 2-1 (3.0) and 1-3.4 (2.0) but that is wrong, it should pick 2-1 and 1-3.
When using 2-1.4 it works correctly:
```
root@z2srv:~# ~mm/uhubctl/uhubctl -l 2-1.4 -p 1 -a off
Current status for hub 2-1.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-1.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 00a0 off
Current status for hub 1-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
Sent power off request
New status for hub 1-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
```

[lsusb.txt](https://github.com/mvp/uhubctl/files/5028629/lsusb.txt)
[uhubctl.txt](https://github.com/mvp/uhubctl/files/5028630/uhubctl.txt)

Thanks a lot for this useful tool!

closed 2020-08-12 by mvp

mvp, 2020-08-12:
Thanks @c-MM , added this hub to the table!

Regarding confused matching of paired hubs - this is existing issue #255.
It is actually manufacturer's fault - USB 3 spec requires hubs to advertise globally unique BOS descriptor container id, which should be the same only on USB2/USB3 pairs. This hub advertises exactly the same BOS container id for all 4 virtual hubs, which makes it difficult to properly classify which one belongs to which pair.
So far, I am unable to come up with universal algorithm that can do that matching without being able to rely on proper BOS container id on all platforms.

vpeter4, 2022-07-28:
I wonder if this USB hub works also without external power supply?

mvp, 2022-07-28:
It may work for very low power downstream devices like mouse or keyboard. But in general it shouldn't - power has to come from somewhere, as upstream hub connection can only provide 5V/1A.

commit 0f6218df (2020-08-12): Add AmazonBasics HUC9003V1EBL 7-port USB 3.1 hub as supported device

Closes #261.