mvp/uhubctl #263: App. get an error on get_hub_info() -> libusb_get_bos_descriptor() of type LIBUSB_ERROR_PIPE

msaparrat1, 2020-08-17:
I debugged your application and libusb_get_bos_descriptor() returns LIBUSB_ERROR_PIPE.
I don't know what is happen, I am lost.
Could you help me?

mvp, 2020-08-17:
* Check that `sudo lsusb -v` can print ContainerID. Normally, you would see something like this (important part here is ` ContainerID             {11234567-89ab-cde0-0123-456789abcdef}`):
```
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x00a9
    Per-port power switching
    Per-port overcurrent protection
    TT think time 16 FS bits
    Port indicators
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent    100 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0303 lowspeed power enable connect
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x002a
  bNumDeviceCaps          3
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000000
      (Missing must-be-set LPM bit!)
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        2047 micro seconds
  Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {11234567-89ab-cde0-0123-456789abcdef}
can't get debug descriptor: Resource temporarily unavailable
```
If that works, `sudo uhubctl` should work without any issues (if you don't want to run `uhubctl` under `sudo`, consider fixing USB permissions as described here https://github.com/mvp/uhubctl#linux-usb-permissions).
* If `sudo lsusb -v` does not print ContainerID for your USB3 hub, try it out on different USB3 device, or on different Linux or Mac host - perhaps it is issue with this particular hub?
* Make sure you are linking to most recent version of libusb package. I am using Ubuntu 20.04, and libusb-1.0.23, and it works fine for me:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```
```
$ dpkg -S /usr/include/libusb-1.0/libusb.h
libusb-1.0-0-dev:amd64: /usr/include/libusb-1.0/libusb.h

$ dpkg -l libusb-1.0-0-dev:amd64
ii  libusb-1.0-0-dev:amd64 2:1.0.23-2build1 amd64        userspace USB programming library development files
```

msaparrat1, 2020-08-18:
Thank you @mvp for you fast response!!
Here I  have attached the output of lsusb -v.
As you can see, there is no information about ContainerId, but the command lsusb in some manner can take the BOD information.
My computer is a Laptop Dell Inspiron 15 7559 I7-6700hq Gtx-960m 16gb.
A view of USB devices is shown below:
[image]
and this:
[image]
[usbsinfo.txt](https://github.com/mvp/uhubctl/files/5087647/usbsinfo.txt)

And the output of your app whit some printf's added:

Info.Vendor: 1d6b:0003
Info.Location: 2
Info.ContainerId: 
Info.Actionable: 0
Info.Level: 1
libusb_get_bos_descriptor() OK!!
Info.Vendor: 1d6b:0003
Info.Location: 2
Info.ContainerId: 
Info.Actionable: 0
Info.Level: 1
libusb_get_bos_descriptor() OK!!
Info.Vendor: 1d6b:0002
Info.Location: 1
Info.ContainerId: 
Info.Actionable: 0
Info.Level: 1
Error intentando obtener BOS descriptor: libusb_get_bos_descriptor() retornó: LIBUSB_ERROR_PIPE
Motivo: Pipe error
Info.Vendor: 1d6b:0002
Info.Location: 1
Info.ContainerId: 
Info.Actionable: 0
Info.Level: 1
Error intentando obtener BOS descriptor: libusb_get_bos_descriptor() retornó: LIBUSB_ERROR_PIPE
Motivo: Pipe error
No compatible smart hubs detected!
Run with -h to get usage info.
There were permission problems while accessing USB.
To fix this, run this tool as root using 'sudo uhubctl',
or add one or more udev rules like below
to file '/etc/udev/rules.d/52-usb.rules':
SUBSYSTEM=="usb", ATTR{idVendor}=="2001", MODE="0666"
then run 'sudo udevadm trigger --attr-match=subsystem=usb'

Best regards.

mvp, 2020-08-18:
> There were permission problems while accessing USB.
> To fix this, run this tool as root using 'sudo uhubctl',
> or add one or more udev rules like below
> to file '/etc/udev/rules.d/52-usb.rules': [...]

This error message clearly suggests that you have to do to fix this, read more here https://github.com/mvp/uhubctl#linux-usb-permissions.

Note that `lsusb -v` will have the same problem - it will not show container ID unless you call it as`sudo lsusb -v`.

If your USB3 device does not advertise container ID (unlikely, but possible), `uhubctl` will still work, but you will have to use 2 commands to turn power off, read more here: https://github.com/mvp/uhubctl#usb-30-duality-note.

closed 2020-08-18 by mvp