mvp/uhubctl #264: Not working in RPi4

jaybauson, 2020-08-24:
I am trying to cut the power in one of my USB port, but it looks like I can only cut the power for a very short time.
The idea is that I would like to catch the temperature in bash using `vcgencmd measure_temp` and when the temperature is higher than 45C, I will turn on the USB to power up my fan. And then turn it off if the target temperature is lower than 35C.
Not sure if this is something doable using uhubctl, it just happened that the fan that I currently have is USB powered. Otherwise, I will use GPIO fan instead.

Here's the information of my RPi4B 8GB

```
root@jaypi:~# vcgencmd bootloader_version
Jul 31 2020 14:43:39
version f8d1dc69144b10b4bd78b4b6d08658ac3005726d (release)
timestamp 1596203019

root@jaypi:~# cat /etc/*release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

root@jaypi:~# uhubctl 
Current status for hub 2 [1d6b:0003 Linux 5.4.51-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0203 power 5gbps U0 enable connect [05e3:0749 Generic USB3.0 Card Reader 000000001532]
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 5.4.51-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]

root@jaypi:~# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 6, If 0, Class=Hub, Driver=hub/4p, 480M
```

I tried the following commands just to make sure I tested the possible combinations to hit the right USB port.
```
uhubctl -a 0 -p 1 -l 2  #- this seems to recycle the power, like a reset. but recycling all USB ports.
uhubctl -a 0 -p 2 -l 2
uhubctl -a 0 -p 3 -l 2
uhubctl -a 0 -p 4 -l 2

uhubctl -a 0 -p 4 -l 1
uhubctl -a 0 -p 3 -l 1
uhubctl -a 0 -p 2 -l 1
uhubctl -a 0 -p 1 -l 1
```

mvp, 2020-08-24:
This is FAQ https://github.com/mvp/uhubctl#power-comes-back-on-after-few-seconds-on-linux

mvp, 2020-08-24:
Also, Raspberry Pi cannot control ports individually. For RPI 4B you should use

uhubctl -l 2 -a 0 -r 1000

Not specifying port number means reset all ports

jaybauson, 2020-08-24:
I see, so either power off all ports or nothing for RPi4?

mvp, 2020-08-24:
Yes, this is limitation of raspberry pi hardware

closed 2020-08-24 by mvp

bucefalo2, 2020-09-13:
> I see, so either power off all ports or nothing for RPi4?

thank you for clarifying ! personally i misunderstood the FAQ part talking about RPI4  and i tried several config but no success !
thank you also to @mvp to confirm !