mvp/uhubctl #266: Pi4, USB-C Port

ganzgustav22, 2020-09-09:
Will (or should) this tool cut and restore power on the Pi4 USB-C Port also?

According to https://github.com/mvp/uhubctl#raspberry-pi-4b
`uhubctl -l 3 -p 1 -a 0`
should cut power. When I type this, I can see the device disappear in dmesg:
```
[  184.239568] dwc2 fe980000.usb: Not connected
[  184.239593] usb 3-1: usbfs: usb_submit_urb returned -19
[  184.239637] dwc2 fe980000.usb: Not connected
[  184.239652] usb 3-1: usbfs: usb_submit_urb returned -19
[  184.315865] usb 3-1: USB disconnect, device number 2

```

When running `uhubctl -l 3 -p 1 -a 1` it doesn't come back though.

ganzgustav22, 2020-09-09:
Well, I've measured with a multimeter on TP1 now (I think this is the testpad that is connected to the USB +5V line?), but the power doesn't get cut off. Also I can measure 0 Ohm between TP1 and the +5V line on the GPIO header, that means it's parallel and the power cannot be controlled via software, or?

mvp, 2020-09-09:
uhubctl can see this device advertising power switching support, and is able to send a request to turn power off.

However, it seems that underlying chip is missing necessary circuitry to actually cut the power on otg line.

I don't know internal details of pi4 otg chip, perhaps @p33m or @slysven know more about pi4 hardware and can shed some light?

P33M, 2020-09-09:
There is no power switching facility on Pi 4's USB-C port.

mvp, 2020-09-09:
Thanks @P33M for confirmation. I will change readme to mention that

closed 2020-09-09 by mvp

ganzgustav22, 2020-09-09:
Thanks for your quick help guys.

commit 3bc22e3c (2020-09-09): Clarify RPi4B support for USB-C OTG port

Closes #266.