mvp/uhubctl #268: Support ganged power switching

henzef, 2020-09-16:
Would it be possible to support USB hubs with ganged power switching in uhubctl? If yes, what would need to be done to implement it (e.g. is it already possible via sysfs or do we need a kernel patch)?

I have several hubs that only support ganged power switching and I would like turn them on and off (not per-port obviously).

mvp, 2020-09-16:
This is duplicate for #260.
Yes, it is possible, but work not started yet. I have some idea how it would be implemented.
No guarantees when it will happen, hopefully soon.

closed 2020-09-16 by mvp

henzef, 2020-09-16:
Actually #260 only mentions reset support. I am aiming for all-port power on/off. Do you have any pointers on how such a feature could be implemented, so I could give it a try?

mvp, 2020-09-16:
It will be command line flag to enable operating on ganged hubs. I have some prototype almost ready

henzef, 2020-09-17:
I managed to switch a D-Link DUB-H4 F-Revision hub on and off with the following hack:

    diff --git a/uhubctl.c b/uhubctl.c
    index 3541fa8..bb79678 100644
    --- a/uhubctl.c
    +++ b/uhubctl.c
    @@ -448,7 +448,7 @@ static int get_hub_info(struct libusb_device *dev, struct hu
    b_info *info)
                 info->ppps = 0;
                 /* Logical Power Switching Mode */
                 int lpsm = uhd->wHubCharacteristics[0] & HUB_CHAR_LPSM;
    -            if (lpsm == HUB_CHAR_COMMON_LPSM && info->nports == 1) {
    +            if (lpsm == HUB_CHAR_COMMON_LPSM) {
                     /* For 1 port hubs, ganged power switching is the same as per-p
    ort: */
                     lpsm = HUB_CHAR_INDV_PORT_LPSM;
                 }