mvp/uhubctl #270: Amazon HU39002V1_ V1_0 2017/11/30

ganzgustav22, 2020-09-28:
I've measured the 5V Pins on the PCB, it seems they are all connected in parallel, i.e. no ganged switching.

Does it work at all, has anybody successfully used it? No matter what I reset, the 5V Pin never goes to 0V. But I'm not sure if I'm doing it correctly.

I've tried `uhubctl -e -l 2-1.4.4 -l 2-1.4 -l 2-1 -l 1-1.1.4.4 -l 1-1.1.4 -l 1-1.1 -a off` as well as several different single hubs and ports, but the power never seems to be disabled.

I'd like to power-cycle this device:
`  Port 3: 1103 power indicator enable connect [0483:5720 Linux 3.10.14 with dwc2-gadget HID Gadget]
`

uhubctl output (with the Raspberry Pi4 internal hubs excluded):
```
Current status for hub 2-1.4.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect

Current status for hub 2-1.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0243 power 5gbps U2 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]

Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0243 power 5gbps U2 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]

Current status for hub 1-1.1.4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power

Current status for hub 1-1.1.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]

Current status for hub 1-1.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 1103 power indicator enable connect [0483:5720 Linux 3.10.14 with dwc2-gadget HID Gadget]
```

[image]

mvp, 2020-09-28:
> I've tried uhubctl -e -l 2-1.4.4 -l 2-1.4 -l 2-1 -l 1-1.1.4.4 -l 1-1.1.4 -l 1-1.1 -a off as well as several different single hubs and ports, but the power never seems to be disabled.

That commanf doesn't make a lot of sense and wouldn't do what you want it to do.
First of all, `-l` cannot be specified multiple times, last usage will be active (I guess I need to emit an error in this case). If you need to control multiple hubs, execute uhubctl multiple times with different parameters.
Second, do not use `-e` unless you know what you are doing. In order to turn power off for USB 3 hubs, you have to turn off power for it's USB 2 compatibility partner. `-e` disables automatic handling for this, it means you have to execute uhubctl twice for both USB 3 and USB 2 hubs (which are actually your hub).
However, for this 10 port hub this gets really complicated, because it's built as 3*4 port hubs, so you will see 6 hubs in total: 3*USB3+3*USB2.
Basically, you may need to hunt for real mapping between USB 2 and USB 3 hubs, and control power carefully, always keeping in mind USB 2/3 duality.

ganzgustav22, 2020-09-28:
How can I identify which usb2 hub belongs to which usb3 hub?

If I don't know exactly which usb2 hub belongs to which usb3 one, could I switch them all off (that's what my intention with the -e and multiple -l parameters was)?

Like this?
```
uhubctl -e -l 2-1.4.4  -a off
uhubctl -e -l 2-1.4 -a off
uhubctl -e -l 2-1 -a off
uhubctl -e -l 1-1.1.4.4 -a off
uhubctl -e -l 1-1.1.4 -a off
uhubctl -e -l 1-1.1 -a off
```

mvp, 2020-09-28:
That should work. Also, it's highly likely that hubs can be correlated by similar USB path, e.g. 2-1.4.4 is likely paired with 1-1.1.4.4. unfortunately, this is not a guarantee on all platforms, otherwise I would have coded it into uhubctl

ganzgustav22, 2020-09-28:
Thanks a lot. That indeed worked, I could measure 0V on the +5V Pin with the above commands.

closed 2020-09-28 by mvp