mvp/uhubctl #272: make file?

vezem, 2020-09-30:
I am VERY new to raspberry pi. I have a RPi4, and was following along fine with the README until it said run the 'make' command to create uhubctl binary. What does that mean? This is what I tried in terminal and this is what I got:

pi@raspberrypi:~ $ make
make: *** No targets specified and no makefile found.  Stop.
pi@raspberrypi:~ $ make uhubctl
make: Nothing to be done for 'uhubctl'.
pi@raspberrypi:~ $ 

It says 'nothing to do done' 

but when I do:
sudo uhubctrl -a 0 -p 1

just to test, it says:
sudo: uhubctl: command not found

Help. I am a huge noob lol

mvp, 2020-09-30:
Your Raspbian or Raspberry Pi OS may already have uhubctl in package feeds, so may already work:
```
sudo apt install uhubctl
```
However, make sure you have version 2.2.0. Anything older does not properly support RPi 4B.

You can also install it with:
```
git clone https://github.com/mvp/uhubctl
cd uhubctl
make
sudo make install
```

closed 2020-09-30 by mvp

seamusdemora, 2020-10-01:
FYI - as of today, the RPi pkg manager will install version 2.0.0 :(

```
$ uhubctl -v
2.0.0
```

Also FWIW - ver 2.0.0 does provide at least some limited support for RPi 4B... I am able to turn the internal hub on and off. At least it **reports** that it has been turned off:

```
$ sudo uhubctl -l 2 -p 2 -a off 
...
$ sudo uhubctl -l 2 -p 2 
Current status for hub 2 [1d6b:0003 Linux 5.4.51-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 2: 0080 off
Current status for hub 1 [1d6b:0002 Linux 5.4.51-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports]
```

mvp, 2020-10-01:
This is a lie. uhubctl version 2.0.0 will not properly turn off voltage, only version 2.2.0 and higher will work properly

seamusdemora, 2020-10-01:
I don't know why you feel the need to call me a liar - I am only reporting what I saw today on my system.

mvp, 2020-10-01:
Haha, I didn't mean to call you a liar 😱
I only meant to say that older version of uhubctl is lying to you 😁 Which is partially my fault.

seamusdemora, 2020-10-01:
Ah - OK... I was about to reply that I have tested this again, and learned that although `uhubctl` **reports** power is off, it actually is still on. I verified this by re-mounting a USB drive that was plugged into the hub that is "powered off" :)

mvp, 2020-10-01:
You can find full list of improvements for each version at https://github.com/mvp/uhubctl/releases. Version 2.0.0 should mostly work ok for USB 2 devices (which is applicable to RPi2/3B/3B+), bit still has some bugs which are fixed in most recent version.