mvp/uhubctl #273: Up Core, how to use it

ivomarvan, 2020-10-01:
I use a small [Up Core](https://up-shop.org/up-core-series.html) computer. It should have "per-port power switching" support. But I can't figure out which command line to use.

Using the 
**`sudo lsusb -v | command grep -i -P "^ Bus | Per-port power switching"`**
I found out they should support it.

Bus 002 Device 003: ID 0bda:8153 Realtek Semiconductor Corp. 
**Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp.**
    **Per-port power switching**
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Bus 001 Device 008: ID 046d:c31c Logitech, Inc. Keyboard K120
**Bus 001 Device 006: ID 0557:8021 ATEN International Co., Ltd CS1764A [CubiQ DVI KVMP Switch]**
    **Per-port power switching**
**Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp.** 
    **Per-port power switching**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub_

For completeness, I also present other forms of the list of my USB ports.
`sudo uhubctl`

_Current status for hub 2-1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [0bda:8153]
Current status for hub 1-2.2 [0557:8021]
  Port 1: 0103 power enable connect [046d:c31c Logitech USB Keyboard]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-2 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.00, 4 ports]
  Port 1: 0100 power
  Port 2: 0103 power enable connect [0557:8021]
  Port 3: 0103 power enable connect [10c4:ea60 Silicon Labs CP2102 USB to UART Bridge Controller 0001]
  Port 4: 0100 power_

`lsusb`

_Bus 002 Device 003: ID 0bda:8153 Realtek Semiconductor Corp. 
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Bus 001 Device 008: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 006: ID 0557:8021 ATEN International Co., Ltd CS1764A [CubiQ DVI KVMP Switch]
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub_

`lsusb -t`

_/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 3, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/7p, 480M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 6, If 0, Class=Hub, Driver=hub/4p, 12M
            |__ Port 1: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 1: Dev 8, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 1: Dev 8, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 7, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M_

**Can you advise me which command line to turn off / cycle a single physical USB port on a UP Core computer?** 

(I know, that it is USB 2 and USB 3:
- Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. and
- Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp.

)

mvp, 2020-10-01:
Just run

```
sudo uhubctl -l {location} -a 0
```
where location is one of 2-1, 1-2 or 1-2.2 (one of listed by uhubctl).
You can try hunting ports one by one by adding `-p {port}`, where port would be 1 to 4.

closed 2020-10-01 by mvp

ivomarvan, 2020-10-02:
Dear

ivomarvan, 2020-10-02:
# @mvp 
Thank you for the quick reply!

I tried it all, but it didn't work.

I created a minimal configuration.  I connected only one device with a control LED to a single USB port of Up Core. Since I couldn't use the keyboard, I controlled the computer via wifi and ssh.
I found that Up Core does not have the necessary support ("No compatible smart hubs found"). The device in my previous experiment that reported the necessary support was probably the KVM switch.

Conclusion: **Used computer _UP Core_ does not support "switching power to one port"**.