mvp/uhubctl #274: DELL P2416D was incorrectly added to the list of supported hubs

maxnikulin, 2020-10-05:
Either there are more revisions of DELL P2416D monitors with different internal hubs, or I have missed something, or this hub should be moved to errata section from the list of supported devices.

From my point of view `uhubctl` is able only to manage data lines, not the power ones. Despite the hub reports per-port power switching, I could not get a LED switched off even when `uhubctl` reported no power at the port. However I could "re-connect" a USB device after `echo 1 >/sys/bus/usb/devices/.../remove` using `uhubctl -a off ...; uhubctl -a on ...` commands.

Actually there is a way to turn off all ports through i2c bus but the screen is turned off as well

```
ddccontrol -r 0xd6 -w 5 dev:/dev/i2c-4  # off
ddccontrol -r 0xe1 -w 0 dev:/dev/i2c-4  # on
```

I am afraid, the author of #164 would not provide more details. In my case it is

- DELL P2416D
- Model: No P2416Db
- Made: week 3 of 2017
- Model in EDID: DELA0C2, DELA0C3
- 4-ports USB 2.0 HUB `0424:2514` (notice that newer models of DELL monitors have USB 3 hubs)

```
Bus 002 Device 057: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
```

```
lsusb -v -d 0424:2514

Bus 002 Device 057: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x0424 Standard Microsystems Corp.
  idProduct          0x2514 USB 2.0 Hub
  bcdDevice            b.b3
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      1 Single TT
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      2 TT per port
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      1 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         0 Full speed (or root) hub
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
  Self Powered
```

Accordingly to README, only hubs that could turn off power lines are considered as supported, so at least a footnote should be added to the entry for DELL P2416D that at least some devices are only partially supported.

### P.S.
An idea that could be discussed in a dedicated issue. Have you considered marking device as not authorized before turning the port off? It seems it could be more reliable than suggesting to perform multiple commands struggling with kernel attempts to restore device state. And it is not so severe as `remove` interface. An example for a mouse connected to the hub in P2416D:

```
# uhubctl -a off -l 2-2 -p 2
Current status for hub 2-2 [0424:2514]
  Port 2: 0303 power lowspeed enable connect [09da:000a A4Tech USB Mouse]
Sent power off request
New status for hub 2-2 [0424:2514]
  Port 2: 0000 off
# uhubctl  -l 2-2 -p 2
Current status for hub 2-2 [0424:2514]
  Port 2: 0303 power lowspeed enable connect [09da:000a A4Tech USB Mouse]

# echo 0 >/sys/bus/usb/devices/2-2.2/authorized
# uhubctl -a off -l 2-2 -p 2
Current status for hub 2-2 [0424:2514]
  Port 2: 0303 power lowspeed enable connect [09da:000a A4Tech USB Mouse]
Sent power off request
New status for hub 2-2 [0424:2514]
  Port 2: 0000 off
# uhubctl  -l 2-2 -p 2
Current status for hub 2-2 [0424:2514]
  Port 2: 0000 off
```

https://www.kernel.org/doc/html/latest/usb/authorization.html
Authorizing (or not) your USB devices to connect to the system

mvp, 2020-10-06:
Thanks for feedback! I will add a note about this monitor, and also put a suggestion to readme about authorization to prevent automatic reconnection. Not sure about coding this Linux-ism into uhubctl itself though...

mvp, 2020-10-06:
Added notes to readme in 2a4df4ee6cd and a7b5f6b362ef.

closed 2020-10-06 by mvp

maxnikulin, 2020-10-07:
Thank you, @mvp. I am not sure as well that maintaining such tricks in code is less burden than explaining to new users why it does not work