mvp/uhubctl #277: Add IC USB2534 to supported list - tested and functional

littlehill, 2020-10-26:
The board i am using has USB2534 and tool helped me a lot.
Reporting another tested piece. All 4 ports.
Regards
---
ID 0424:2534 Standard Microsystems Corp.
..
wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection

mvp, 2020-10-27:
Thanks for reporting!

Can you please confirm vbus off support per https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl?

seamusdemora, 2020-10-27:
@mvp: Support for *per-port-switching* is a decision by the designer/manufacturer obviously, but is *per-port-switching* supported by all USB chipsets - or just a subset? In other words, if a designer/mfr chose the "wrong" chipset, would that take away the option for providing *per-port-switching* as a feature? Could the driver be a factor in this?

mvp, 2020-10-27:
@seamusdemora, actually this is off topic for this issue. That said, to the best of my knowledge, all USB chipsets do support per-port power switching. However, they can be configured by manufacturer to disable such support to match real hardware capabilities. Some vendors configure it improperly - e.g. advertise PPPS support when actual circuitry is missing.

mvp, 2020-10-29:
@littlehill, can you please confirm if vbus off is supported for this hub per https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl?

littlehill, 2020-10-29:
> @littlehill, can you please confirm if vbus off is supported for this hub per https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl?

Greetings, yes i can confirm that the IC functionality is there.

mvp, 2020-10-29:
@littlehill , my question is not whether chip reports power switching, but if voltage is actually turning off for your hub.

littlehill, 2020-10-30:
Greetings, yes, thats what i mean, i see the voltage switching. On per-port basis. Was able to hook osciloscope on and in both cases with & without a device connected it does switch the voltage as expected.
Sorry for not being clear about this from start.
The hub has EN/OC functionality in 1 pin, so power switch AP2553A is used. (Just basically a transistor with OC limit)
The hub controlls the behaviour and switches power on/off. Measured on HW.

mvp, 2020-11-01:
Thanks @littlehill, added this hub to the list in 77d8851.

commit 77d8851c (2020-11-01): Add Microchip EVB-USB2534BC as supported device

Closes #277.
closed 2020-11-01 by mvp