mvp/uhubctl #282: Different behavior on M1 (Apple Silicon) Mac

baoshan, 2020-11-19:
On Intel-based Mac, the `cycle` action:

1. power off the port
2. wait (configurable delay)
3. turn it on
4. exit 

The system can detect the device was removed during step 2.

On M1 (Apple Silicon) Mac (Mac Mini tested), the system can only detect the device was removed after step 4, not during step 2. I use a USB audio device and the system sound panel to demonstrate the issue:

[image]

mvp, 2020-11-19:
Please show output of `uhubctl`. You have USB 2.1 devices shown, which suggests you must have USB 3.x that are their dual partners. In order to turn off power properly, both have to be turned off. There is a possibility that uhubctl is not able to automatically detect those dual devices.

Also, note that M1 silicon is very new. I didn't know of it's existence just a few days ago. I wouldn't be surprised if it had some bugs that prevent uhubctl from working properly. Thanks!

baoshan, 2020-11-19:
Thanks for the reply. I think `uhubctl` cycles the port properly in above screenshot, but before the command exists, not in the delay.

The out of `uhubctl` is:

```
baoshan@Baoshans-Mini ~ % uhubctl
Current status for hub 1-1.4 [043e:9a48]
  Port 1: 0503 power highspeed enable connect [043e:9a42 LG Electronics Inc. USB Audio]
  Port 2: 0301 power lowspeed connect []
  Port 3: 0100 power
  Port 4: 0103 power enable connect [043e:9a40 LG Electronics Inc. USB Controls]
Current status for hub 1-1 [043e:9a46 LG Electronics Inc. USB2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0103 power enable connect [262a:100c FiiO FiiO USB DAC K1]
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [043e:9a48]
```

Please let me know if further diagnosis is required.

mvp, 2020-11-19:
Few things:

1. I find it really odd that you don't see USB 3.x devices as dual partners for `hub 1-1 [043e:9a46 LG Electronics Inc. USB2.1 Hub, USB 2.10, 4 ports]`. This might be weird feature of new M1 silicon.

2. Your output shows devices without [vid:pid]. This suggests that you are using libusb-1.0.23 which has a bug on Mac OS. You need to install libusb from master branch to fix this, please follow FAQ here https://github.com/mvp/uhubctl#compiling for workaround for Mac.

3. I wouldn't exclude possibility that to really fix this, we might need libusb adjusted for MacOS 11 (Big Sur). Main suspect is M1 USB 3 internal devices now showing up in uhubctl output.

baoshan, 2020-11-19:
1. For comparison, below is the output for Intel-based Mac (same configuration, the hub is an Ultrafine 4K monitor):

```
Current status for hub 20-2.4 [043e:9a48]
  Port 1: 0503 power highspeed enable connect [043e:9a42 LG Electronics Inc. USB Audio]
  Port 2: 0301 power lowspeed connect []
  Port 3: 0100 power
  Port 4: 0103 power enable connect [043e:9a40 LG Electronics Inc. USB Controls]
Current status for hub 20-2 [043e:9a46 LG Electronics Inc. USB2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0103 power enable connect [262a:100c FiiO FiiO USB DAC K1]
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [043e:9a48]
```

2. I was using `1.0.22` of `libusb` and now `HEAD` of both `libusb` and `uhubctl`. Still different behavior.

mvp, 2020-11-19:
libusb-1.0.22 or 1.0.24 (head right now) should be ok, only 1.0.23 was broken. I am talking about this device with empty brackets:
```
  Port 2: 0301 power lowspeed connect []
```

I still find it odd that your USB 2.1 devices don't show their USB 3.x counterparts. Are you filtering full uhubctl output?

baoshan, 2020-11-19:
The first is on Intel-based Mac:

[image]

The second is on M1:

[image]

mvp, 2020-11-19:
Thank you for screenshots.

I don't know what to say about this except for perhaps different behavior of darwin 20.x. Note that Linux currently has similar buggy behavior under certain conditions - notifications of powering USB off is not always properly propagated to user space (udev).

We may need to get more experience with new M1 hardware, which is available to very few people right now...

baoshan, 2020-11-19:
Thanks for your comment. 

macOS 11 (Big Sur) on Intel-based Macs does exhibit the expected behavior. Only M1 has unexpected behavior. For people facing the same issue, manually `off`, `sleep(2)`, then `on` should temporarily mitigate the issue.

Thanks again and please let me know if I can further help.

mvp, 2020-11-19:
That's extremely odd and should not be possible. Note that
```
uhubctl -a cycle -d {delay}
```
is exactly equivalent to:
```
uhubctl -a off
sleep {delay}
uhubctl -a on
```

baoshan, 2020-11-19:
I’m reporting another behavior which is could be the culprit:

After an `off` command, the device takes about 30 seconds to disappear, this is abnormal.
During that period, an `on` command will make the device disappear immediately and then appear shortly.

mvp, 2020-11-19:
That could be either issue with darwin 20, or with libusb not adapted to darwin 20 differences on M1. All that uhubctl does is sending low level control message to the hub kernel driver to turn power off (and on).

baoshan, 2020-11-19:
I guess it’s probably the issue with darwin 20, since VBUS did drop after the `off` command, which proves `libusb` is not the one to suspect?

SlySven, 2020-11-19:
> Also, note that M1 silicon is very new. I didn't know of it's existence just a few days ago. I wouldn't be surprised if it had some bugs that prevent uhubctl from working properly. Thanks!

:rofl: 

Apparently that is not the only problem:
https://www.slashgear.com/apple-silicon-m1-macs-become-unusable-when-restoring-macos-17647819/

:man_facepalming:

baoshan, 2020-11-20:
> > Also, note that M1 silicon is very new. I didn't know of it's existence just a few days ago. I wouldn't be surprised if it had some bugs that prevent uhubctl from working properly. Thanks!
> 
> 🤣
> 
> Apparently that is not the only problem:
> https://www.slashgear.com/apple-silicon-m1-macs-become-unusable-when-restoring-macos-17647819/
> 
> 🤦‍♂️

The `Erase Mac` option in the Recovery utility menu works for me. Give it a try.

seamusdemora, 2020-11-20:
This is a completely off-topic comment, so please delete it if it's a bother, but: 

Since buying the latest Macbook Pro about a year ago (T2 chip, Catalina, etc), I have become hugely put off by Apple's "new approach" to computing. Their priorities, and their decisions, leave me cold. Their attitude seems to be that we plebes will consume whatever they serve, and they owe us no explanations - or even current or consistent documentation. I won't be buying anything from Apple unless their house is put back in order.

mvp, 2020-12-01:
@baoshan, can you please try building libusb for M1?

```
export ARCHFLAGS="-arch arm64"
brew install automake autoconf libtool
brew uninstall --ignore-dependencies libusb
brew install -s --HEAD libusb
brew uninstall uhubctl
brew tap mvp/uhubctl https://github.com/mvp/uhubctl
brew install -s --HEAD uhubctl
```
This should give you `uhubctl` that works natively for M1.

baoshan, 2020-12-02:
It builds and executes fine. 

```
baoshan@Baoshans-Mini ~ % which uhubctl
/usr/local/bin/uhubctl

baoshan@Baoshans-Mini ~ % ls -al /usr/local/bin/uhubctl
lrwxr-xr-x  1 baoshan  admin  53 Dec  2 13:09 /usr/local/bin/uhubctl -> /opt/homebrew/Cellar/uhubctl/HEAD-0e733b6/bin/uhubctl

baoshan@Baoshans-Mini ~ % uhubctl -h
uhubctl e733b69: utility to control USB port power for smart hubs.
Usage: uhubctl [options]
Without options, show status for all smart hubs.

baoshan@Baoshans-Mini ~ % uhubctl
Current status for hub 1-1.4 [043e:9a48]
  Port 1: 0503 power highspeed enable connect [043e:9a42 LG Electronics Inc. USB Audio]
  Port 2: 0301 power lowspeed connect []
  Port 3: 0100 power
  Port 4: 0103 power enable connect [043e:9a40 LG Electronics Inc. USB Controls]
Current status for hub 1-1 [043e:9a46 LG Electronics Inc. USB2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [262a:100c FiiO FiiO USB DAC K1]
  Port 4: 0503 power highspeed enable connect [043e:9a48]
```

Please let me know if there’s something else I could help.

The behavior specific to this issue is unchanged.
> After an off command, VBUS drops immediately, but the (audio) device takes ~30 seconds to disappear from the System Preferences panel.

mvp, 2020-12-02:
Thank you @baoshan, it's good to know that uhubctl can be built natively for M1.

Since power does drop immediately, it confirms that uhubctl is working as expected. There is nothing I could do to change MacOS behavior to remove device from device tree immediately. So, I am going to close this issue as not actionable. Please feel free to reopen if you don't agree. Much thanks for reporting!

closed 2020-12-02 by mvp

baoshan, 2020-12-02:
Thanks! Will report back once the situation changes.

mcuee, 2020-12-06:
From #292, uhubctl works fine under my Mac Mini M1. My USB Webcam comes and goes with the power control (using PhotoBooth for testing).

```
mcuee@mcuees-Mac-mini uhubctl % lsusb -t
  Bus 000.Dev 001: USB 3.1 Bus, 
    |__ Bus 002.Dev 026: 4-Port USB 2.0 Hub, 480Mb/s
        |__ Bus 002.Dev 028: 4-Port USB 2.0 Hub, 480Mb/s
            |__ Bus 002.Dev 032: USB3.0 Card Reader, 480Mb/s
        |__ Bus 002.Dev 008: PROLiNK PCC3220, 480Mb/s
    |__ Bus 002.Dev 001: USB 2.0 Hub [MTT], 480Mb/s
        |__ Bus 002.Dev 006: USB Receiver, 12Mb/s
        |__ Bus 002.Dev 010: USB 2.0 Hub [MTT], 480Mb/s
            |__ Bus 002.Dev 013: PICkit 2 Microcontroller Programmer, 12Mb/s
            |__ Bus 002.Dev 012: Vendor-Specific Device, 480Mb/s
            |__ Bus 002.Dev 014: SES084D, 480Mb/s
            |__ Bus 002.Dev 011: ZY Control Mic, 12Mb/s
        |__ Bus 002.Dev 005: BCM20702A0, 12Mb/s
    |__ Bus 002.Dev 027: 4-Port USB 3.0 Hub, 5Gb/s
        |__ Bus 002.Dev 030: 4-Port USB 3.0 Hub, 5Gb/s

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl_m1                    
Current status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0280 power 5gbps SS.Disabled
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1.1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [05e3:0749 Generic USB3.0 Card Reader 000000001536]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 2-5 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0ac8:3420 Vimicro Corp. PROLiNK PCC3220]

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl_m1  -a off -p 4 -l 2-1
Current status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 00a0 off
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0503 power highspeed enable connect [0ac8:3420 Vimicro Corp. PROLiNK PCC3220]
Sent power off request
New status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0000 off

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl_m1                    
Current status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0280 power 5gbps SS.Disabled
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 00a0 off
Current status for hub 2-1.1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [05e3:0749 Generic USB3.0 Card Reader 000000001536]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 2-5 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0000 off

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl_m1  -a on -p 4 -l 2-1 
Current status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 00a0 off
Sent power on request
New status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0000 off
Sent power on request
New status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0101 power connect [0ac8:3420]
```

mcuee, 2020-12-06:
So far the only problem is with USB HID device. I tested hidapi and sending random invalid requests with crash the Mac Mini M1 (automatically reboot). Previously this does not happen with my dead Mac Mini 2011 (died in early 2019).

My native M1 build of homebrew. You can see I have USB related apps and all seem to work.
mcuee@mcuees-Mac-mini ~ % which brew
/opt/homebrew/bin/brew
mcuee@mcuees-Mac-mini ~ % brew list 
autoconf	confuse		hidapi		libtool		minicom		picocom		sdcc		swig
automake	gdbm		icu4c		libusb		open-ocd	pkg-config	sphinx-doc	usb.ids
boost		gettext		libftdi		libusb-compat	openssl@1.1	python@3.9	sqlite		xz
cmake		gputils		libhid		lsusb		pcre		readline	stlink

mvp, 2020-12-06:
No, uhubctl does NOT work properly on M1 for you. This command `uhubctl_m1  -a off -p 4 -l 2-1` should turn off power for hubs 2-1 and 2-5, but it matches 2-5.1, which is incorrect. I have started work to fix this in #288 - can you please give it a try?

mcuee, 2020-12-06:
I see. Thanks for pointing this out.
#288 seems to work.

```
mcuee@mcuees-Mac-mini test % git clone https://github.com/mvp/uhubctl -b usb3-duality
mcuee@mcuees-Mac-mini test % cd uhubctl

mcuee@mcuees-Mac-mini uhubctl % CFLAGS="-I/opt/homebrew/include" LDFLAGS="-L/opt/homebrew/lib -lusb-1.0" make
cc  -I/opt/homebrew/include -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"2.2.0-19-g0de9e546\" uhubctl.c -o uhubctl -L/opt/homebrew/lib -lusb-1.0 -lusb-1.0

mcuee@mcuees-Mac-mini uhubctl % file uhubctl
uhubctl: Mach-O 64-bit executable arm64

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl   
Current status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0280 power 5gbps SS.Disabled
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1.1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [05e3:0749 Generic USB3.0 Card Reader 000000001536]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 2-5 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0ac8:3420 Vimicro Corp. PROLiNK PCC3220]

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl -a off -p 4 -l 2-1
Current status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 00a0 off
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0503 power highspeed enable connect [0ac8:3420 Vimicro Corp. PROLiNK PCC3220]
Sent power off request
New status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0000 off
```

mvp, 2020-12-06:
Nope, matching is still incorrect. But thanks for testing it, I will adjust #288 to deal with this

mcuee, 2020-12-06:
You are right. Sorry but if I only want to match 2-1.4 PROLiNK PCC3220, which command should I use?

mvp, 2020-12-06:
You can disable USB3 duality handling by adding -e, but then you have to execute uhubctl twice: once for USB2, and second for appropriate USB3 partner hub:

    uhubctl -a off -p 4 -l 2-1 -e
    uhubctl -a off -p 4 -l 2-5 -e

mcuee, 2020-12-06:
I see. Thanks.

```

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl -a on -p 4 -l 2-5 -e
Current status for hub 2-5 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 00a0 off
Sent power on request
New status for hub 2-5 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 02a0 power 5gbps Rx.Detect
mcuee@mcuees-Mac-mini uhubctl % ./uhubctl -a on -p 4 -l 2-1 -e
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0000 off
Sent power on request
New status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0101 power connect [0ac8:3420]
mcuee@mcuees-Mac-mini uhubctl % ./uhubctl -a off -p 4 -l 2-5 -e
Current status for hub 2-5 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-5 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 00a0 off
mcuee@mcuees-Mac-mini uhubctl % ./uhubctl -a off -p 4 -l 2-1 -e
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0503 power highspeed enable connect [0ac8:3420 Vimicro Corp. PROLiNK PCC3220]
Sent power off request
New status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0000 off
```