mvp/uhubctl #283: Device power does not turn off + LED lamp not recognized regardless

slimjet, 2020-11-21:
I want to turn on and off to several simple USB lights with uhubctl and it's not working at present. For one, when I run uhbuctl, the devices do not seem to be recognized. I purchased the USB3-HUB7C Plugable 7 port USB port because it was registered as working with uhubctl. I am on Ubuntu.

When I run uhubctl, I can see the USB mouse I've connected and my phone, but neither of the two USB lights:
```
Current status for hub 2-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [1a7c:0191 Kingsis Peripherals Evoluent VerticalMouse 4]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [18d1:4ee1 Google Pixel 3a 94GAY0P1KG]
Current status for hub 2-2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```
When I turn off the ports via uhubctl, power turns off for the mouse and the phone, but stays on for the usb lights:
```
sudo uhubctl -a off -p 1-4 -l 2-2.1
Current status for hub 2-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [1a7c:0191 Kingsis Peripherals Evoluent VerticalMouse 4]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [18d1:4ee1 Google Pixel 3a 94GAY0P1KG]
Sent power off request
New status for hub 2-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
```
So. I'm not sure if this is something that can be fixed by trying a different USB hub, or if the USB lights themselves just draw so little power that it's going to be a problem no matter what. They're dumb USB powered LED lights, and I assume they won't be recognized. I'm willing to try purchasing different USB lights if that will solve the problem, and trying different USB hubs if that will make uhubctl work, but I want to see what the community makes of my problem before I go off buying a bunch of junk on Amazon. Any insight?

mvp, 2020-11-21:
There are two possible issues:

 One is if your hub is USB 3,  and requires powering off both USB 2 and 3 components (I somehow don't see USB 3 in your list, which is possible if you don't use compliant cable or don't have proper USB 3 upstream connection) - see a note about it in README.md.

Another possibility is that this hub may not really support VBUS off. This Information about hubs supporting VBUS (voltage) off is submitted by other users. Sometimes people don't properly test VBUS off, and other times vendors change actual circuitry between different hardware releases. If you are sure your hub doesn't support VBUS off (do a test per readme), then I would like to remove it from supported list. All other hubs in the list should support VBUS off properly, in particular all Amazon basics do.

mvp, 2020-11-26:
@slimjet, can you please perform VBUS test off for your USB3-HUB7C Plugable 7 port hub?
If you confirm that it does not support VBUS off, I will remove it from compatible list.
Thanks!

mvp, 2020-12-01:
Closing for inactivity from submitter. Please feel free to reopen if you have anything to add. Thank you!

closed 2020-12-01 by mvp

slimjet, 2020-12-12:
Alright, have some updates. I switched to my macbook pro and installed uhubctl using Homebrew. I'm very new at this, but so far as I can tell I followed the directions to install the non-broken version.

The plugable hub (and another, more on that in a moment) only works with uhubctl on two of the 7 ports. The dumb usb light will turn off on port 1 and 7, which happen to be the two ports that have a little lighhtning bolt underneath the port, (I assume indicating that they are for charging devices faster). I don't know why only two ports will work to be turned off with uhubctl. That's not good enough for my use case.

Here's the terminal output for it:

> Joshuas-MacBook-Pro:~ joshuah$ uhubctl -a off -p 1-4 -l 20-2.1
> Current status for hub 20-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0103 power enable connect [1a7c:0191 Kingsis Peripherals Evoluent VerticalMouse 4]
>   Port 2: 0100 power
>   Port 3: 0100 power
>   Port 4: 0100 power
> Sent power off request
> New status for hub 20-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Current status for hub 1-2.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 02a0 power 5gbps Rx.Detect
>   Port 2: 02a0 power 5gbps Rx.Detect
>   Port 3: 02a0 power 5gbps Rx.Detect
>   Port 4: 02a0 power 5gbps Rx.Detect
> Sent power off request
> New status for hub 1-2.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off
> Joshuas-MacBook-Pro:~ joshuah$ uhubctl -a off -p 1-4 -l 1-2.1
> Current status for hub 20-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Sent power off request
> New status for hub 20-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Current status for hub 1-2.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off
> Sent power off request
> New status for hub 1-2.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off

Another weird element of this is that the hub has small lights, one for each port that's activate (eg turns on when I plug in a mouse or a usb drive, etc). They don't light up when I plug in a usb light, but that has no effect on whether the hub will turn on or off the power to the ports.

I'm experiencing basically the same problem with another USB hub. This is the Amazon basics "USB 3.0 7-port HUB with AC Adapter", b07v2bgm4r.

> Joshuas-MacBook-Pro:~ joshuah$ uhubctl -a off -p 1-4 -l 20-2
> Current status for hub 20-2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0503 power highspeed enable connect [1a40:0101 USB 2.0 Hub]
>   Port 2: 0103 power enable connect [1a7c:0191 Kingsis Peripherals Evoluent VerticalMouse 4]
>   Port 3: 0100 power
>   Port 4: 0100 power
> Sent power off request
> New status for hub 20-2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Current status for hub 1-2 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 02a0 power 5gbps Rx.Detect
>   Port 2: 02a0 power 5gbps Rx.Detect
>   Port 3: 02a0 power 5gbps Rx.Detect
>   Port 4: 02a0 power 5gbps Rx.Detect
> Sent power off request
> New status for hub 1-2 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off
> Joshuas-MacBook-Pro:~ joshuah$ uhubctl -a off -p 1-4 -l 1-2
> Current status for hub 20-2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Sent power off request
> New status for hub 20-2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Current status for hub 1-2 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off
> Sent power off request
> New status for hub 1-2 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off

One and ONLY one port turns off using uhubctl. As with the plugable, it's the first port, which is blue, that turns off. None of the others actually turn off.

The fact that this behavior is so similar to the plugable leads me to wonder if something is up with how I install uhubctl, or maybe with my macbook pro? It seems odd that I would have basically the same problems with two hubs, where uhubctl really only works on one or two ports.