mvp/uhubctl #284: Static binary?

DainBB, 2020-11-23:
Hi,

Is there a chance of i386 and x64 statically linked binary or instruction how to build one? 

I'd like to try uhubctl on QNAP x64 NAS but it has a very old libusb so the only option seems to build statically linked binary but I'm not having much luck with it.

I'm using Centos 8 and, installed glibc-static and compiled static version of libusb1 to have libusb-1.0.a but when I'm trying compile uhubctl as following I'm getting gazillion messages about undefined symbols in libpthread, and rebuilding that does not seem like a fun idea.

# cc -static -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"2.2.0-15-gdb8c4a59\" uhubctl.c -o uhubctl -Wl,-zrelro,-znow -lusb-1.0
/usr/lib/gcc/x86_64-redhat-linux/8/libgcc_eh.a(unwind-dw2-fde-dip.o): In function `__register_frame_info_bases.part.5':
(.text+0x16b8): undefined reference to `pthread_mutex_lock'
/usr/lib/gcc/x86_64-redhat-linux/8/libgcc_eh.a(unwind-dw2-fde-dip.o): In function `__register_frame_info_table_bases':
(.text+0x17cc): undefined reference to `pthread_mutex_lock'
/usr/lib/gcc/x86_64-redhat-linux/8/libgcc_eh.a(unwind-dw2-fde-dip.o): In function `__deregister_frame_info_bases':
(.text+0x1892): undefined reference to `pthread_mutex_lock'
(.text+0x1915): undefined reference to `pthread_mutex_unlock'
(.text+0x194d): undefined reference to `pthread_mutex_unlock'
/usr/lib/gcc/x86_64-redhat-linux/8/libgcc_eh.a(unwind-dw2-fde-dip.o): In function `_Unwind_Find_FDE':
(.text+0x1a84): undefined reference to `pthread_mutex_lock'
(.text+0x1bb8): undefined reference to `pthread_mutex_unlock'
(.text+0x1bf9): undefined reference to `pthread_mutex_unlock'
/usr/lib/gcc/x86_64-redhat-linux/8/libgcc_eh.a(unwind-dw2-fde-dip.o): In function `__register_frame_info_bases.part.5':
(.text+0x16eb): undefined reference to `pthread_mutex_unlock'
/usr/lib/gcc/x86_64-redhat-linux/8/libgcc_eh.a(unwind-dw2-fde-dip.o): In function `__register_frame_info_table_bases':
(.text+0x17ff): undefined reference to `pthread_mutex_unlock'
/usr/lib/gcc/x86_64-redhat-linux/8/../../../../lib64/libusb-1.0.a(libusb_1_0_la-core.o): In function `usbi_mutex_lock':
/root/libusb-1.0.23/libusb/os/threads_posix.h:47: undefined reference to `pthread_mutex_lock'
/usr/lib/gcc/x86_64-redhat-linux/8/../../../../lib64/libusb-1.0.a(libusb_1_0_la-core.o): In function `usbi_mutex_unlock':
/root/libusb-1.0.23/libusb/os/threads_posix.h:51: undefined reference to `pthread_mutex_unlock'
/usr/lib/gcc/x86_64-redhat-linux/8/../../../../lib64/libusb-1.0.a(libusb_1_0_la-core.o): In function `usbi_mutex_init':
/root/libusb-1.0.23/libusb/os/threads_posix.h:43: undefined reference to `pthread_mutex_init'
/usr/lib/gcc/x86_64-redhat-linux/8/../../../../lib64/libusb-1.0.a(libusb_1_0_la-core.o): In function `usbi_mutex_lock':
/root/libusb-1.0.23/libusb/os/threads_posix.h:47: undefined reference to `pthread_mutex_lock'
/usr/lib/gcc/x86_64-redhat-linux/8/../../../../lib64/libusb-1.0.a(libusb_1_0_la-core.o): In function `usbi_mutex_unlock':
/root/libusb-1.0.23/libusb/os/threads_posix.h:51: undefined reference to `pthread_mutex_unlock'
/usr/lib/gcc/x86_64-redhat-linux/8/../../../../lib64/libusb-1.0.a(libusb_1_0_la-core.o): In function `usbi_mutex_lock':
/root/libusb-1.0.23/libusb/os/threads_posix.h:47: undefined reference to `pthread_mutex_lock'

[and so on]

Thank you in advance!

mvp, 2020-11-23:
I don't have an answer for statically linking with libusb - if you figure this out, please let me know.
This is perhaps better asked at https://github.com/libusb/libusb/issues or in libusb mailing list.

However, speaking of very old libusb versions and dynamically linking uhubctl to them. Note that uhubctl should work reasonably well with USB2 devices with libusb as old as 1.0.12 (released in 2012), and to work well with USB3 devices I recommend libusb 1.0.16 or higher.

DainBB, 2020-11-23:
Thanks, I'll keep digging and will share result if I get any success. 

And when I say really old libusb on QNAP, it is really-really old

[admin@QNAP lib]# ls -l libusb*
lrwxrwxrwx 1 admin administrators    19 2020-10-29 01:51 libusb-0.1.so.4 -> libusb-0.1.so.4.4.4
-rw-r--r-- 1 admin administrators 20112 2020-09-29 14:38 libusb-0.1.so.4.4.4
lrwxrwxrwx 1 admin administrators    19 2020-10-29 01:51 libusb-1.0.so -> libusb-1.0.so.0.0.0*
lrwxrwxrwx 1 admin administrators    19 2020-10-28 15:52 libusb-1.0.so.0 -> libusb-1.0.so.0.1.0*
-rwxr-xr-x 1 admin administrators 54696 2020-09-29 14:38 libusb-1.0.so.0.0.0*
lrwxrwxrwx 1 admin administrators    50 2020-10-28 15:52 libusb-1.0.so.0.1.0 -> /mnt/ext/opt/mtpBinary/usr/lib/libusb-1.0.so.0.1.0*
lrwxrwxrwx 1 admin administrators    19 2020-10-29 01:51 libusb.so -> libusb-0.1.so.4.4.4

mvp, 2020-11-23:
Another thought - rather than trying to create static binary, perhaps you could link uhubctl to custom newer version of `libusb.so` using [this recipe](https://stackoverflow.com/a/8835402/1734130)?

DainBB, 2020-11-23:
That was a great idea, so what I did is below, just in case someone else needs it, it's all applicable to x64 version of QNAP of course.

Installed libusb1-devel and gcc on Centos 7 with glibc 2.17, Centos 8 is not good as glibc version there is 2.28 and on QNAP it's 2.21, so binaries compiled on Centos 8 won't work on QNAP.
Compiled uhubctl as usual
Copied some libraries from to /opt/usr/lib and created symlinks:

[admin@QNAP lib]# ls -la /opt/usr/lib
total 1076
drwxr-xr-x 2 admin administrators    340 2020-11-23 18:02 ./
drwxr-xr-x 3 admin administrators     60 2020-11-23 17:56 ../
lrwxrwxrwx 1 admin administrators     14 2020-11-23 18:00 libcap.so.2 -> libcap.so.2.22*
-rwxr-xr-x 1 admin administrators  20048 2020-11-23 17:59 libcap.so.2.22*
-rwxr-xr-x 1 admin administrators 338672 2020-11-23 17:59 libdw-0.176.so*
lrwxrwxrwx 1 admin administrators     14 2020-11-23 18:00 libdw.so.1 -> libdw-0.176.so*
-rwxr-xr-x 1 admin administrators  99952 2020-11-23 18:01 libelf-0.176.so*
lrwxrwxrwx 1 admin administrators     15 2020-11-23 18:01 libelf.so.1 -> libelf-0.176.so*
lrwxrwxrwx 1 admin administrators     16 2020-11-23 18:02 liblzma.so.5 -> liblzma.so.5.2.2*
-rwxr-xr-x 1 admin administrators 157400 2020-11-23 18:01 liblzma.so.5.2.2*
lrwxrwxrwx 1 admin administrators     17 2020-11-23 17:57 libmount.so.1 -> libmount.so.1.1.0*
-rwxr-xr-x 1 admin administrators 277808 2020-11-23 17:56 libmount.so.1.1.0*
lrwxrwxrwx 1 admin administrators     16 2020-11-23 17:57 libudev.so.1 -> libudev.so.1.6.2*
-rwxr-xr-x 1 admin administrators  91024 2020-11-23 17:56 libudev.so.1.6.2*
lrwxrwxrwx 1 admin administrators     19 2020-11-23 17:56 libusb-1.0.so -> libusb-1.0.so.0.1.0*
lrwxrwxrwx 1 admin administrators     19 2020-11-23 17:57 libusb-1.0.so.0 -> libusb-1.0.so.0.1.0*
-rwxr-xr-x 1 admin administrators 103680 2020-11-23 17:56 libusb-1.0.so.0.1.0*

Export new libraries path
[admin@QNAP lib]#export LD_LIBRARY_PATH=/opt/./usr/lib/

Just to verify no libraries missing:
[admin@QNAP lib]# ldd /opt/uhubctl
        linux-vdso.so.1 (0x00007fffa81e4000)
        libusb-1.0.so.0 => /opt/./usr/lib/libusb-1.0.so.0 (0x00007f543c602000)
        libc.so.6 => /lib/libc.so.6 (0x00007f543c25f000)
        libudev.so.1 => /opt/./usr/lib/libudev.so.1 (0x00007f543c049000)
        libpthread.so.0 => /lib/libpthread.so.0 (0x00007f543be2c000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f543c81b000)
        librt.so.1 => /lib/librt.so.1 (0x00007f543bc24000)
        libcap.so.2 => /opt/./usr/lib/libcap.so.2 (0x00007f543ba1f000)
        libm.so.6 => /lib/libm.so.6 (0x00007f543b71a000)
        libdw.so.1 => /opt/./usr/lib/libdw.so.1 (0x00007f543b4c9000)
        libdl.so.2 => /lib/libdl.so.2 (0x00007f543b2c5000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x00007f543b0af000)
        libattr.so.1 => /usr/lib/libattr.so.1 (0x00007f543aeab000)
        libelf.so.1 => /opt/./usr/lib/libelf.so.1 (0x00007f543ac93000)
        libz.so.1 => /lib/libz.so.1 (0x00007f543aa79000)
        liblzma.so.5 => /opt/./usr/lib/liblzma.so.5 (0x00007f543a853000)
        libbz2.so.1 => /usr/lib/libbz2.so.1 (0x00007f543a643000)

[admin@QNAP lib]# lsusb
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 1005:b155 Apacer Technology, Inc.
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 007: ID 0bc2:331a Seagate RSS LLC
Bus 004 Device 008: ID 0bc2:ac35 Seagate RSS LLC
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

[admin@QNAP lib]# /opt/uhubctl
Current status for hub 2-1 [8087:0024]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1 [8087:0024]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [1005:b155 USB DISK MODULE 19DC1A00AC037673]
  Port 4: 0100 power

Now It looks like USB hub I have here is Intel Q87 PCH and it's most certainly does not advertise any any per-power capabilities but it's something to start playing with.

Thanks a lot!

mvp, 2020-11-23:
Glad it worked for you! As issue is resolved, I am closing it :-)

closed 2020-11-23 by mvp