mvp/uhubctl #285: Only turn off data for USB port

JuliusHenke, 2020-11-24:
Thanks for this cool tool, but I am missing the option to have my GoPro camera provided with power but not data.

Could you add an optional flag to only turn off data on a certain USB port but still provide it with voltage (leave VBUS on). Is this even possible?

mvp, 2020-11-24:
1. If your hub does not support per-port power switching properly, uhubctl will cut off data connection but not power.

2. If your hub does support per-port power switching properly, it is NOT possible to turn off only data but not VBUS power - per USB specs, ceasing USB connection must kill power too. Or to explain it differently: e.g you see these connection features in uhubctl output:

```Port 3: 0503 power highspeed enable connect
```
`power`, `highspeed`, `enable`, `connect` are connection features. Some of them can be enabled or disabled, e.g. `power` feature (and this is what uhubctl is doing to control power). However few of those features are implicit, and cannot be enabled or disabled, e.g. `connect` feature. If we were able to disable `connect` feature but keep `power`, that would do exactly what you want. However, USB protocol does not provide such ability.

closed 2020-11-24 by mvp

JuliusHenke, 2020-11-24:
Ok, thanks for you answer.
My hub supports per-port power switching and also turns off VBUS power on the port. I opened a pull request to include my hub in the README: #286