mvp/uhubctl #290: List of compatible vs Incompatible hubs

raveslave, 2020-12-01:
Would be nice with more recent hubs added to the list.   
Especially those that are easy to get from companies like Anker, TP-Link etc.

Just tried the TP-Link, UH700 and seems it doesn't work. It reports port is off, but power was still there.

mvp, 2020-12-01:
This was discussed before - unfortunately, maintaining incompatible list is not sustainable. Such list would be 10x larger than compatible list, I don't have bandwidth that.

closed 2020-12-01 by mvp

spidgorny, 2020-12-08:
Community will help you to maintain the list (separate .md file or wiki).
I just bought a hub, which says it has the "Per-port power switching", but it doesn't work.
I would save a lot of time and money if I could see other people report that this hub is not compatible.
Please consider this black-list again.

"RSHTECH Aluminium USB Hub Active 3.0 with 36 W (12 V / 3 A) Power Supply 7 Port USB 3.0 Hub with Quick Charge (4 USB 3.0 Data Transfer Port, 3 Quick Charge Ports) with Single Switch, RSH-A37"

mvp, 2020-12-08:
Sorry, but maintaining incompatible list is a bit too much to handle - it will be 10-100x times larger than list of compatible hubs - I cannot boil the ocean.

Now, talking about your hub. Are you sure it does not work? How did you test it?
I actually have RSHTECH 16 port hub (thanks to RSHTECH team for sending me one!), and seems to work fine.
Can you try building uhubctl from master branch?
Then follow https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl to test if it works. Thanks!

spidgorny, 2020-12-08:
Tried to turn off everything. It says: "OK", but the devices attached still working. Tested on MacBook and Ras Pi.
```
➜  uhubctl git:(master) ./uhubctl -p 1-4 -a 0 
Current status for hub 20-17 [0bda:0411 Generic 2-Port USB 3.0 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
  Port 2: 00a0 off
  Port 3: 00a0 off
  Port 4: 00a0 off
Sent power off request
New status for hub 20-17 [0bda:0411 Generic 2-Port USB 3.0 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
  Port 2: 00a0 off
  Port 3: 00a0 off
  Port 4: 00a0 off
Current status for hub 20-5 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Sent power off request
New status for hub 20-5 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
➜  uhubctl git:(master) ./uhubctl                    
Current status for hub 20-17 [0bda:0411 Generic 2-Port USB 3.0 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
  Port 2: 00a0 off
  Port 3: 00a0 off
  Port 4: 00a0 off
Current status for hub 20-5 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
```

mvp, 2020-12-08:
Since you have 7 port hub, you should see 4 hubs total: 2 usb2 + 2 usb3.
Can you post output of `uhubctl -f`?

spidgorny, 2020-12-08:
```
➜  uhubctl git:(master) ./uhubctl -f
Current status for hub 20-17 [0bda:0411 Generic 2-Port USB 3.0 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-5 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

spidgorny, 2020-12-08:
The other three ports are just Quick Charge - no data, just power.

mvp, 2020-12-08:
When you say connected devices are still working, do you mean even data connection is working? Or connected phone continues to charge?

Just to make sure, I tested VBUS off on RSHTECH RSH-A16, and it does work as expected - turning port off makes connected phone to stop charging, and it loses data connection too.

spidgorny, 2020-12-08:
I have attached a fan, USB light and a phone. All of them keep running/shining/charging. I don't care about the data TX.

mvp, 2020-12-08:
Ah, ok. I guess RSH-A37 is not working properly. Might be worth sending request to RSHTECH team, they are quite responsive guys.

mvp, 2020-12-14:
@spidgorny, I mentioned your issue with RSH-A37 in email communication with RSHTECH. They said that chip used in this hub is different due to 3 qc ports, and that might be the reason it doesn't work for you. It definitely works with RSH-A16, and I would expect it should work with RSW-A37S (very similar to RSH-A37, but 7 data ports).

spidgorny, 2020-12-14:
Thank you for taking care of my trouble. I have been able to find D-Link DUB-H4 on ebay. It works.

justinclift, 2020-12-24:
@raveslave @spidgorny As a general thought, since @mvp doesn't really want to maintain a list of "known incompatible" hubs, maybe one of you could do it in your own repo?

If the list turns out to be useful and maintainable, then @mvp could add a link to it from here and not have to be worried about the maintenance headache. :smile: