mvp/uhubctl #292: To use pkg-config for macOS

mcuee, 2020-12-06:
I think it is a good idea to use pkg-config for macOS as well. Both Homebrew and Macports support pkg-config.

For example, with the new Apple Silicon macOS devices, Homebrew recommends the native install to be at "/opt/homebrew" and not the default "/usr/local".

```
mcuee@mcuees-Mac-mini uhubctl % CFLAGS="-I/opt/homebrew/include" LDFLAGS="-L/opt/homebrew/lib -lusb-1.0" make
cc  -I/opt/homebrew/include -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"2.2.0-20-g0e733b69\" uhubctl.c -o uhubctl -L/opt/homebrew/lib -lusb-1.0 -lusb-1.0

mcuee@mcuees-Mac-mini uhubctl % pkg-config --libs libusb-1.0
-L/opt/homebrew/Cellar/libusb/1.0.23/lib -lusb-1.0

mcuee@mcuees-Mac-mini uhubctl % pkg-config --cflags libusb-1.0
-I/opt/homebrew/Cellar/libusb/1.0.23/include/libusb-1.0
```

mvp, 2020-12-06:
I love this idea, but on my Macbook Pro 16" and latest homebrew pkg-config is not available by default.
If homebrew had pkg-config installed by default, I would completely agree.

mcuee, 2020-12-06:
I see what you mean. Is it possible to check the availability of pkg-config. If yes then use it. If not then hard-code.

mvp, 2020-12-06:
Not sure, I will think about it. If you have an idea how to implement this in Makefile, I would love to know

mcuee, 2020-12-06:
Quick fix. Please check if this is okay.

```
diff --git a/Makefile b/Makefile
index f1b242a..4c9a7fb 100644
--- a/Makefile
+++ b/Makefile
@@ -25,6 +25,10 @@ ifeq ($(UNAME_S),Linux)
 endif
 
 ifeq ($(UNAME_S),Darwin)
+ifneq (,$(shell which pkg-config))
+    CFLAGS  += $(shell pkg-config --cflags libusb-1.0)
+    LDFLAGS += $(shell pkg-config --libs libusb-1.0)
+endif
 ifneq ($(wildcard /opt/local/include),)
        # MacPorts
        CFLAGS  += -I/opt/local/include
```

mvp, 2020-12-06:
Looks good except that you need to use tabs, not spaces. Let me put a fix for this right now, I see spaces in other lines of Makefile, that's a bug :-)

mvp, 2020-12-06:
Thanks for raising this issue @mcuee , I have committed universal pkg-config fix for all platforms.
It still works for Linux and Mac homebrew if pkg-config is not installed.

closed 2020-12-06 by mvp

mcuee, 2020-12-06:
Great. Latest git works perfectly under my Mac Mini M1.

mvp, 2020-12-06:
Out of curiosity, can you please provide output of `uhubctl` with any USB3 hub attached to M1?
I wonder if there is some discrepancy of USB stack on M1 hardware as was reported in #282.

mcuee, 2020-12-06:
Here you are.

```

mcuee@mcuees-Mac-mini uhubctl % lsusb
Bus 002 Device 023: ID 0bda:0411 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub 
Bus 002 Device 025: ID 0bda:0411 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub 
Bus 002 Device 027: ID 05e3:0749 Genesys Logic, Inc. USB3.0 Card Reader  Serial: 000000001536
Bus 002 Device 001: ID 1a40:0201 TERMINUS TECHNOLOGY INC. USB 2.0 Hub [MTT] 
Bus 002 Device 003: ID 0a5c:21e8 Broadcom Corp. BCM20702A0  Serial: 5CF3708A6D48
Bus 002 Device 002: ID 1a40:0201 TERMINUS TECHNOLOGY INC. USB 2.0 Hub [MTT] 
Bus 002 Device 007: ID 1915:1025 Nordic Semiconductor ASA ZY Control Mic 
Bus 002 Device 006: ID 0e8d:1836 MediaTek Inc. SES084D  Serial: R96X157
Bus 002 Device 005: ID 04b4:8613 Cypress Semiconductor Vendor-Specific Device 
Bus 002 Device 008: ID 04d8:0033 Microchip Technology Inc. PICkit 2 Microcontroller Programmer  Serial: pk2new
Bus 002 Device 004: ID 046d:c52b Logitech Inc. USB Receiver 
Bus 002 Device 022: ID 0bda:5411 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub 
Bus 002 Device 026: ID 0ac8:3420 Z-Star Microelectronics Corporation PROLiNK PCC3220 
Bus 002 Device 024: ID 0bda:5411 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub 
Bus 000 Device 001: ID 1d6b:XHCI XHCI
1100 Linux Foundation USB 3.1 Bus 

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl
Current status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [05e3:0749 Generic USB3.0 Card Reader 000000001536]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1.1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 2-5 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0ac8:3420 Vimicro Corp. PROLiNK PCC3220]```

mcuee, 2020-12-06:
If I unplug the TP-Link UH700 3.0 Hub, then the output is totally different.

```

mcuee@mcuees-Mac-mini uhubctl % lsusb
Bus 002 Device 001: ID 1a40:0201 TERMINUS TECHNOLOGY INC. USB 2.0 Hub [MTT] 
Bus 002 Device 003: ID 0a5c:21e8 Broadcom Corp. BCM20702A0  Serial: 5CF3708A6D48
Bus 002 Device 002: ID 1a40:0201 TERMINUS TECHNOLOGY INC. USB 2.0 Hub [MTT] 
Bus 002 Device 007: ID 1915:1025 Nordic Semiconductor ASA ZY Control Mic 
Bus 002 Device 006: ID 0e8d:1836 MediaTek Inc. SES084D  Serial: R96X157
Bus 002 Device 005: ID 04b4:8613 Cypress Semiconductor Vendor-Specific Device 
Bus 002 Device 008: ID 04d8:0033 Microchip Technology Inc. PICkit 2 Microcontroller Programmer  Serial: pk2new
Bus 002 Device 004: ID 046d:c52b Logitech Inc. USB Receiver 
Bus 000 Device 001: ID 1d6b:XHCI XHCI
1100 Linux Foundation USB 3.1 Bus 

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl
No compatible smart hubs detected!
Run with -h to get usage info.

```

mvp, 2020-12-06:
That is very interesting!!!

USB3 hubs advertise themselves as 2 apparently independent devices: USB2 and USB3.
On all other platforms, they are always attached to different USB busses, e.g. this is one of my hubs attached to x86 Mac (USB2 bus 20, and USB3 bus 1):
```
Current status for hub 1-1.1 [0bda:0411 Generic 4-Port USB 3.1 Hub, USB 3.20, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-1.1 [0bda:5411 Generic 4-Port USB 2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```
and this is one of my hubs on Linux (USB2 bus 3, USB3 bus 4):
```
Current status for hub 4-3 [174c:3074 Asmedia ASM107x 00000000A001, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 3-3 [174c:2074 Asmedia ASM107x 00000000A001, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 3: 0100 power
```

However, on M1, both USB2 and USB3 components belong to the same bus 2!
I will have to tweak USB2 vs USB3 matching code now, unless this is libusb bug :-(

mcuee, 2020-12-06:
This is the output from libusb examples under Mac Mini M1 (native M1 build).

```

mcuee@mcuees-Mac-mini libusb % ./examples/listdevs 
05e3:0749 (bus 2, device 11) path: 5.1.1
0ac8:3420 (bus 2, device 32) path: 1.4
0bda:0411 (bus 2, device 31) path: 5.1
0bda:5411 (bus 2, device 30) path: 1.1
0bda:0411 (bus 2, device 29) path: 5
0bda:5411 (bus 2, device 28) path: 1
04d8:0033 (bus 2, device 8) path: 2.5.1
1915:1025 (bus 2, device 7) path: 2.5.6
0e8d:1836 (bus 2, device 6) path: 2.5.4
04b4:8613 (bus 2, device 5) path: 2.5.2
046d:c52b (bus 2, device 4) path: 2.2
0a5c:21e8 (bus 2, device 3) path: 2.7
1a40:0201 (bus 2, device 2) path: 2.5
1a40:0201 (bus 2, device 1) path: 2
mcuee@mcuees-Mac-mini libusb % ./examples/testlibusb 
Dev (bus 2, device 11): 05E3 - 0749 speed: 5G
  Manufacturer:              Generic
  Product:                   USB3.0 Card Reader
Dev (bus 2, device 32): 0AC8 - 3420 speed: 480M
  Manufacturer:              Vimicro Corp.
  Product:                   PROLiNK PCC3220
Dev (bus 2, device 31): 0BDA - 0411 speed: 5G
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub
Dev (bus 2, device 30): 0BDA - 5411 speed: 480M
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
Dev (bus 2, device 29): 0BDA - 0411 speed: 5G
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub
Dev (bus 2, device 28): 0BDA - 5411 speed: 480M
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
Dev (bus 2, device 8): 04D8 - 0033 speed: 12M
  Manufacturer:              Microchip Technology Inc.
  Product:                   PICkit 2 Microcontroller Programmer
Dev (bus 2, device 7): 1915 - 1025 speed: 12M
  Manufacturer:              ZY.Ltd
  Product:                   ZY Control Mic
Dev (bus 2, device 6): 0E8D - 1836 speed: 480M
  Manufacturer:              TSST Inc    
  Product:                   SES084D
Dev (bus 2, device 5): 04B4 - 8613 speed: 480M
Dev (bus 2, device 4): 046D - C52B speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 2, device 3): 0A5C - 21E8 speed: 12M
  Manufacturer:              Broadcom Corp
  Product:                   BCM20702A0
Dev (bus 2, device 2): 1A40 - 0201 speed: 480M
  Product:                   USB 2.0 Hub [MTT]
Dev (bus 2, device 1): 1A40 - 0201 speed: 480M
  Product:                   USB 2.0 Hub [MTT]
```

mvp, 2020-12-06:
uhubctl is using following notation for device id: b-x.y.z, where b is bus bus number as returned by `libusb_get_bus_number`, and x,y,z are port numbers for all hubs in chain as returned by `libusb_get_port_path`.
As far as I know, bus numbers for USB2 vs USB3 device should be always different.
But, in your `testlibusb` output all devices belong to bus 2, that's weird!

mcuee, 2020-12-06:
I see. Let me check in libusb mailing list if this is indeed a problem with libusb or a quirk in M1 Macs.

mvp, 2020-12-06:
I think M1 always lists all USB devices as belonging to  bus 2. If libusb just ignored this root device completely, and used next level port as bus number on M1, everything will start behaving normally.

mcuee, 2020-12-06:
In any case, it seems uhubctl works fine under my Mac Mini. I used it to control the power to my USB Web Cam and it seems to work fine.

```
mcuee@mcuees-Mac-mini uhubctl % lsusb -t
  Bus 000.Dev 001: USB 3.1 Bus, 
    |__ Bus 002.Dev 026: 4-Port USB 2.0 Hub, 480Mb/s
        |__ Bus 002.Dev 028: 4-Port USB 2.0 Hub, 480Mb/s
            |__ Bus 002.Dev 032: USB3.0 Card Reader, 480Mb/s
        |__ Bus 002.Dev 008: PROLiNK PCC3220, 480Mb/s
    |__ Bus 002.Dev 001: USB 2.0 Hub [MTT], 480Mb/s
        |__ Bus 002.Dev 006: USB Receiver, 12Mb/s
        |__ Bus 002.Dev 010: USB 2.0 Hub [MTT], 480Mb/s
            |__ Bus 002.Dev 013: PICkit 2 Microcontroller Programmer, 12Mb/s
            |__ Bus 002.Dev 012: Vendor-Specific Device, 480Mb/s
            |__ Bus 002.Dev 014: SES084D, 480Mb/s
            |__ Bus 002.Dev 011: ZY Control Mic, 12Mb/s
        |__ Bus 002.Dev 005: BCM20702A0, 12Mb/s
    |__ Bus 002.Dev 027: 4-Port USB 3.0 Hub, 5Gb/s
        |__ Bus 002.Dev 030: 4-Port USB 3.0 Hub, 5Gb/s

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl_m1  -a on -p 4 -l 2-1
Current status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 00a0 off
Sent power on request
New status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0000 off
Sent power on request
New status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0101 power connect [0ac8:3420]

mcuee@mcuees-Mac-mini uhubctl % ./uhubctl_m1  -a off -p 4 -l 2-1
Current status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-5.1 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports]
  Port 4: 00a0 off
Current status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0503 power highspeed enable connect [0ac8:3420 Vimicro Corp. PROLiNK PCC3220]
Sent power off request
New status for hub 2-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0000 off
```

```

commit d3a79ace (2020-12-06): Use pkg-config when available

Remove custom handling for different platforms for CFLAGS and LDFLAGS
and prefer using pkg-config when available.
However, this is tested to still build ok on Linux on Mac howebrew
even if pkg-config is not available.

Closes #292, should also simplify issues raised by #142, #147, #289 and 5db248771eb8192.