mvp/uhubctl #295: Updates on plugable behavior + amazon basics behavior

slimjet, 2020-12-12:
Alright, have some updates. I switched to my macbook pro and installed uhubctl using Homebrew. I'm very new at this, but so far as I can tell I followed the directions to install the non-broken version.

The plugable hub (and another, more on that in a moment) only works with uhubctl on two of the 7 ports. The dumb usb light will turn off on port 1 and 7, which happen to be the two ports that have a little lighhtning bolt underneath the port, (I assume indicating that they are for charging devices faster). I don't know why only two ports will work to be turned off with uhubctl. That's not good enough for my use case.

Here's the terminal output for it:

> Joshuas-MacBook-Pro:~ joshuah$ uhubctl -a off -p 1-4 -l 20-2.1
> Current status for hub 20-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0103 power enable connect [1a7c:0191 Kingsis Peripherals Evoluent VerticalMouse 4]
>   Port 2: 0100 power
>   Port 3: 0100 power
>   Port 4: 0100 power
> Sent power off request
> New status for hub 20-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Current status for hub 1-2.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 02a0 power 5gbps Rx.Detect
>   Port 2: 02a0 power 5gbps Rx.Detect
>   Port 3: 02a0 power 5gbps Rx.Detect
>   Port 4: 02a0 power 5gbps Rx.Detect
> Sent power off request
> New status for hub 1-2.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off
> Joshuas-MacBook-Pro:~ joshuah$ uhubctl -a off -p 1-4 -l 1-2.1
> Current status for hub 20-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Sent power off request
> New status for hub 20-2.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Current status for hub 1-2.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off
> Sent power off request
> New status for hub 1-2.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off

Another weird element of this is that the hub has small lights, one for each port that's activate (eg turns on when I plug in a mouse or a usb drive, etc). They don't light up when I plug in a usb light, but that has no effect on whether the hub will turn on or off the power to the ports.

I'm experiencing basically the same problem with another USB hub. This is the Amazon basics "USB 3.0 7-port HUB with AC Adapter", b07v2bgm4r.

> Joshuas-MacBook-Pro:~ joshuah$ uhubctl -a off -p 1-4 -l 20-2
> Current status for hub 20-2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0503 power highspeed enable connect [1a40:0101 USB 2.0 Hub]
>   Port 2: 0103 power enable connect [1a7c:0191 Kingsis Peripherals Evoluent VerticalMouse 4]
>   Port 3: 0100 power
>   Port 4: 0100 power
> Sent power off request
> New status for hub 20-2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Current status for hub 1-2 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 02a0 power 5gbps Rx.Detect
>   Port 2: 02a0 power 5gbps Rx.Detect
>   Port 3: 02a0 power 5gbps Rx.Detect
>   Port 4: 02a0 power 5gbps Rx.Detect
> Sent power off request
> New status for hub 1-2 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off
> Joshuas-MacBook-Pro:~ joshuah$ uhubctl -a off -p 1-4 -l 1-2
> Current status for hub 20-2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Sent power off request
> New status for hub 20-2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
>   Port 1: 0000 off
>   Port 2: 0000 off
>   Port 3: 0000 off
>   Port 4: 0000 off
> Current status for hub 1-2 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off
> Sent power off request
> New status for hub 1-2 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
>   Port 1: 0080 off
>   Port 2: 0080 off
>   Port 3: 0080 off
>   Port 4: 0080 off

One and ONLY one port turns off using uhubctl. As with the plugable, it's the first port, which is blue, that turns off. None of the others actually turn off.

The fact that this behavior is so similar to the plugable leads me to wonder if something is up with how I install uhubctl, or maybe with my macbook pro? It seems odd that I would have basically the same problems with two hubs, where uhubctl really only works on one or two ports.

_Originally posted by @slimjet in https://github.com/mvp/uhubctl/issues/283#issuecomment-743555878_

mvp, 2020-12-12:
Can you please tell which exact model of Amazon Basics hub you have?

slimjet, 2020-12-12:
yeah: [b07v2bgm4r](https://www.amazon.com/gp/product/B07V2BGM4R/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1)

mvp, 2020-12-12:
Sorry, but this hub is not in supported list either. There are hundreds and perhaps thousands different models out there, and only few of them are supporting per port power switching properly

mvp, 2020-12-12:
Anyway, thank you for updates, but uncompliant hardware is not something that uhubctl can fix. We will only list hubs that actually work.

Creating and maintaining list of unsupported hubs is not feasible, as it will probably be 10x larger than list of working hubs.

closed 2020-12-12 by mvp