mvp/uhubctl #301: Doesn't work on Raspberry pi 4B

Spyman, 2020-12-22:
After update doesn't work on Raspberry pi 4B. 
"sudo uhubctl" output: 
Current status for hub 2 [1d6b:0003 Linux 5.4.79-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1 [1d6b:0002 Linux 5.4.79-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  
  All ports enabled. 
  "uhubctl -a 0" has no effect. 
  
"uhubctl -l 1-1 -a 0" output: 
No compatible smart hubs detected at location 1-1!
Run with -h to get usage info.
There were permission problems while accessing USB.
To fix this, run this tool as root using 'sudo uhubctl',
or add one or more udev rules like below
to file '/etc/udev/rules.d/52-usb.rules':
SUBSYSTEM=="usb", ATTR{idVendor}=="2001", MODE="0666"
then run 'sudo udevadm trigger --attr-match=subsystem=usb'

mvp, 2020-12-22:
Make sure you use latest version 2.3.0. Your output is not consistent with that version

mvp, 2020-12-22:
After you get version 2.3.0 installed/compiled, please post output of `sudo uhubctl`

Spyman, 2020-12-22:
Thank you.
Not the latest version was in the binary repository.
Everything works in the sources compiled from the gita.

closed 2020-12-22 by Spyman