mvp/uhubctl #302: quiet mode

cainbantam, 2020-12-29:
Hello,
thanks for the great tool. Would it be possible to add a silent mode, for I've heard that this is a requirement if we want cron to be able to use it (I have a made a script using the uhubctl command which fails to be run by cron and this might be the reason why...)
Thank you for your work!

mvp, 2020-12-29:
It should work from cron already, few issues you should pay attention to:

* Be sure to specify full path to uhubctl executable in your cron invocation, e.g. `/usr/sbin/uhubctl`
* Be sure that you have either configured udev USB permissions as described in readme, or run uhubctl as root (it should be already the case if you run it from root's crontab)
* By default, cron will send email with output of called executable. You can mute it by either assigning MAILTO="" at the beginning of crontab, or by redirecting stdout/stderr to `/dev/null`, e.g.

    uhubctl -l 1-1 -p 2 -a 0 >/dev/null 2>&1

closed 2020-12-29 by mvp

cainbantam, 2020-12-29:
great! Thank you for those further advices!