mvp/uhubctl #303: EXSYS EX-1113HMS and EX-1184HMV full supported

goski89, 2021-01-05:
Full support for per-port power switching for both models. 
If anyone interested, I recommend this products, industrial quality, still commercially available, maybe high price but worth it.

mvp, 2021-01-05:
Thanks for reporting. Please check if VBUS off is supported - all too often this doesn't work.

mvp, 2021-01-08:
@goski89, please confirm if these hubs actually support VBUS off per https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl:

Plug a phone, USB light or USB fan into USB port of your hub. Try using uhubctl to turn power off on that port, and check that phone stops charging, USB light stops shining or USB fan stops spinning. If VBUS doesn't turn off, your hub manufacturer did not include circuitry to actually cut power off. Such hub would still work to cut off USB data connection, but it cannot turn off power, and we do not consider this supported device.

goski89, 2021-01-08:
@mvp Actually I am on holiday :) I give you an answer when I back to work.

mvp, 2021-01-14:
Ping! Can you please check VBUS off support for this hub?

goski89, 2021-01-14:
Ok, I made some additional tests and I have few conslusions:
1) uhubctl doesn't work on Rpi4 with both models, I don't know why
2) 1184HMV doesnt support VBUS Off
3) 1113HMS support VBUS Off, but like I say, only on Rpi3B+

mvp, 2021-01-14:
Make sure you are testing with uhubctl version 2.3.0 or later (e.g. build it from master branch) - it has USB 3 fixes necessary to make most USB 3 hubs work properly with Raspberry Pi 4B.

Alternatively, you can test USB 3 hubs by connecting them to another Linux or Mac - but make sure you have native USB 3 connection. Easiest way to check if you have USB 3 connection is to see your hub in uhubctl output 2 or more times - refer to readme note about USB 3.0.

mvp, 2021-01-15:
Added ExSys EX-1113HMS in 6ac502a.

closed 2021-01-15 by mvp

goski89, 2021-01-18:
I checked again and EXSYS EX-1113HMS works great on RPI4 (full with VBUS disabled), but EX-1184HMV only disconnects data lines. We find a problem with the hardware. 1113 has Genesis chipset inside but 1184 has QL811 chipset which supports VBUS Off but in this configuration the power switch IC is not mounted on the PCB, there is a fuse bridge. So QL811 cannot switch VBUS even though it supports this functionality. Now we try to find a PS IC chip that will match the PCB layout and then we check if our fix will work.

commit 6ac502a0 (2021-01-15): Add ExSys EX-1113HMS as compatible device

Closes #303.