mvp/uhubctl #306: No compatible devices detected on Raspberry Pi 3 B+ on NetBSD 9.1

rockyhotas, 2021-01-14:
Hello!
I just cloned this repository on NetBSD 9.1, installed on a Raspberry Pi 3 B+, following the steps we defined on a [previous issue](https://github.com/mvp/uhubctl/issues/147#issuecomment-515403273). I ran `gmake`, which exited normally, creating the executable `uhubctl`. However, if I run as `root`

    # ./uhubctl

no compatible devices are detected. Raspberry Pi 3 B+ [should have them](https://github.com/mvp/uhubctl/#compatible-usb-hubs), instead.
What could it be the problem?

mvp, 2021-01-14:
Can you please show output of `uhubctl -f`?

rockyhotas, 2021-01-14:
The same as without options:

```
# ./uhubctl
No compatible devices detected!
Run with -h to get usage info.
# ./uhubctl -f
No compatible devices detected!
Run with -h to get usage info.
```

If it can be useful:

```
# ./uhubctl -v
2.3.0-6-gf76dc986
```

mvp, 2021-01-14:
`uhubctl -f` should list all hubs, whether they support power switching or not.
Is there `lsusb` equivalent in NetBSD? Can it list what you expect?
This could be libusb issue on NetBSD. Which libusb version you have installed? `1.0.22`or later is recommended.
I would recommend debugging it by adding `printf()` in `usb_find_hubs()`, somewhere around here: https://github.com/mvp/uhubctl/blob/f76dc9865325efb5c22806cd49c532fbf47ba233/uhubctl.c#L701

It could be bailing on bad return code from `get_hub_info()`: https://github.com/mvp/uhubctl/blob/f76dc9865325efb5c22806cd49c532fbf47ba233/uhubctl.c#L707

rockyhotas, 2021-01-15:
> `uhubctl -f` should list all hubs, whether they support power switching or not.
> Is there `lsusb` equivalent in NetBSD? Can it list what you expect?

There's no `lsusb` or similar AFAIK, but the `dmesg` should provide all we need:

```
[     2.081312] dwctwo0: Core Release: 2.80a (snpsid=4f54280a)
[     2.081312] usb0 at dwctwo0: USB revision 2.0
...
[     2.101183] uhub0 at usb0: NetBSD (0000) DWC2 root hub (0000), class
9/0, rev 2.00/1.00, addr 1
[     2.111182] uhub0: 1 port with 1 removable, self powered
...
[     3.963921] uhub1 at uhub0 port 1: vendor 0424 (0x424) product 2514 (0x2514), class 9/0, rev 2.00/b.b3, addr 2
[     3.973922] uhub1: multiple transaction translators
[     3.983924] uhub1: 4 ports with 3 removable, self powered
[     5.284044] uhub2 at uhub1 port 1: vendor 0424 (0x424) product 2514 (0x2514), class 9/0, rev 2.00/b.b3, addr 3
[     5.294046] uhub2: multiple transaction translators
[     5.304048] uhub2: 3 ports with 2 removable, self powered
[     5.634091] uhub0: autoconfiguration error: illegal enable change, port 1
...
[     7.054290] mue0 at uhub2 port 1
```

`uhub1` and `uhub2` should be respectively the "Main hub" and "secondary hub" of [your description](https://github.com/mvp/uhubctl/#raspberry-pi-3b). `mue0` is the Ethernet card, in port 1 as expected.

Maybe this error is the reason for this issue?

```
uhub0: autoconfiguration error: illegal enable change, port 1
```

If I connect a mouse and a keyboard to these USB ports they work.

> This could be libusb issue on NetBSD. Which libusb version you have installed? `1.0.22`or later is recommended.

It's `1.0.24`.

> I would recommend debugging it by adding `printf()` in `usb_find_hubs()`, somewhere around here:
> 
> https://github.com/mvp/uhubctl/blob/f76dc9865325efb5c22806cd49c532fbf47ba233/uhubctl.c#L701

Ok, thanks for the position. I tried:
```
        rc = libusb_get_device_descriptor(dev, &desc);
        /* only scan for hubs: */
        printf("Scanning for hubs...\n");
```
But nothing changes in the output:
```
# ./uhubctl
No compatible devices detected!
Run with -h to get usage info.
# ./hubctl -f
No compatible devices detected!
Run with -h to get usage info.
```
I am not familiar with the source. If there's anything else that I should print, let me know.

mvp, 2021-01-15:
The only way for this to happen if this call to `libusb_get_device_list()` returns success, but
`usb_devs` contains empty list. Perhaps something is wrong with libusb on NetBSD?

https://github.com/mvp/uhubctl/blob/f76dc9865325efb5c22806cd49c532fbf47ba233/uhubctl.c#L996

You can try to validate if listdevs from libusb examples is working:

    git clone https://github.com/libusb/libusb
    cd libusb
    # build libusb
    ./autogen.sh
    ./configure
    make
    # build examples
    cd examples
    make
    # run simple libusb utilities
    ./listdevs
    ./testlibusb

If `listdevs` shows nothing, there is nothing uhubctl can do to fix.

rockyhotas, 2021-01-15:
> If `listdevs` shows nothing, there is nothing uhubctl can do to fix.

Thank you so much for the very detailed steps. I followed all of them, and nor `listdevs`, neither `testlibusb` give an output. They all complete successfully, but with no results. From the `root` prompt:

# ./listdevs
# echo $?
0
# ./testlibusb
# echo $?
0

So, maybe it's as you guessed, there is a problem with `libusb` on NetBSD in this case.

mvp, 2021-01-15:
@rockyhotas, I recommend filing issue with libusb https://github.com/libusb/libusb/issues/new - refer to details you did above in new ticket.
I will close this one as there is nothing I can do to fix it :-(

closed 2021-01-15 by mvp

rockyhotas, 2021-01-16:
Ok! Of course, I understand. Thanks for all your help!