mvp/uhubctl #307: Not all D-Link silver DUB-H4 rev B can do power switching

danfos, 2021-01-18:
Wanted to have a hub that supported power switching and traced down a second hand D-Link silver DUB-H4

The label at the bottom says "H/W Ver: B7" and with the README.md saying "rev B" I thought worth a try but testing it shows power switching does not work and opening the box, the PCB shows quite some components missing. The VID:PID is not 05E3:0605 but 05E3:0608

Would be good to add this info to the table so others do not by such a second hand hub.

mvp, 2021-01-19:
Thanks for information! I will update readme, perhaps tomorrow

danfos, 2021-01-20:
To be more precise, the hub does not support VBUS (voltage) off support, it still does cut off USB data connection but README.md clearly says:

> If VBUS doesn't turn off, your hub manufacturer did not include circuitry to actually cut power off. Such hub would still work to cut off USB data connection, but it cannot turn off power, and we do not consider this supported device.

commit fa449eac (2021-01-21): Add a note that D-Link DUB-H4 rev B7+ is not supported

Closes #307.
closed 2021-01-21 by mvp