mvp/uhubctl #308: Hagibis dock (for Mac mini)

trevoz, 2021-01-19:
https://www.amazon.com.au/Enclosure-Hagibis-Docking-Station-MacBook/dp/B088B4DCF2/

$ uhubctl
Current status for hub 0-1 [1d5c:5001 Fresco Logic, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect  
  Port 4: 0203 power 5gbps U0 enable connect [174c:55aa asmedia ASMT1051 12345678F7F5]
Current status for hub 20-2 [1d5c:5011 Fresco Logic, Inc. USB2.0 Hub, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [1a40:0101 USB 2.0 Hub, USB 2.00, 4 ports, ganged]
  Port 4: 0100 power
  Port 5: 0100 power

Hub 0-1 (should this be 20-1?) can have power turned on/off independently of Hub 20-2.

uhubctl -l 20-2 -p 5 -a off - just acts on the USB 2.0 virtual hub

uhubctl -l 0-1 -e -p 4 -a off - just acts on the USB 3.1 virtual hub

mvp, 2021-01-19:
1. Hub locations look like b-x.y.z, where b is USB bus, and x, y, z, ... are port numbers for all hubs in chain, starting from root hub. For this dock, USB 3 hub is reported as attached port 1 of root hub on bus 0, and USB 2 as attached to port 2 of bus 20. Why is that? Ask Apple :-)
2. In order to control power properly on USB 3 hubs, it is absolutely necessary to power off both USB 2 and USB 3 ports of the same physical hub. Normally, uhubctl will automatically detect dual USB 2/3 hubs and switch power on both. However, for this dock, somehow USB 3 hub reports 4 ports, whereas USB 2 dual device has 5 ports. This confuses uhubctl automatic matching logic, making it think that they are completely unrelated devices, and matching cannot function. It means you have to issue 2 uhubctl commands (one for USB 3 and another for USB 2, but using the same port number). In this case, you don't have to use -e - it only makes difference if you want to explicitly disable automatic duality matching code for USB 3 hubs.
3. If you want to add this dock to uhubctl supported devices table, we need to make sure that it supports turning off VBUS. Please follow FAQ item on how to do it https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl. Note that you will need to use 2 uhubctl commands for your dock to reliably test it, e.g. 

        uhubctl -l 0-1 -p 1 -a 0
        uhubctl -l 20-2 -p 1 -a 0

If this turns off VBUS (e.g. phone connected to port 1 stops charging), we will add this dock to supported list. Thanks!

trevoz, 2021-01-19:
OK will do the recommended tests tomorrow.

BTW, the -e was needed to only turn off power to port 4 of the USB 3.1 virtual hub otherwise it turned power off to both the USB 3.1 and USB 2.1 hubs according to uhubctl - I have verified that it does turn on/off the power to the USB 3.1 port 4 as that is where my internal SSD is located (the whole reason for my using uhubctl is to turn the power back on for the SSD after the dock and/or Crucial MX500 SSD has gone to sleep and disappeared never to be heard from  again without re-plugging/rebooting).

mvp, 2021-01-21:
@trevoz, did you have chance to test VBUS off for all ports of this dock?

trevoz, 2021-01-21:
I tested with a USB speaker with power LED - it does not turn off the power, just the data lines  :( So not eligible to be added to the list.

mvp, 2021-01-21:
Thanks for effort, thats unfortunate! I will close this then...

closed 2021-01-21 by mvp