mvp/uhubctl #309: Raspberry pi 4b USB led light not turning off

ChristiaanG, 2021-01-19:
Hello I am using a raspberry pi 4 model b and I want to use it to turn the USB light in the picture on and off.

[image]

The light is connected to port 1 but when I run uhubctl it only shows my mouse and keyboard.

Current status for hub 2 [1d6b:0003 Linux 5.4.79-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [04d9:0180 USB-HID Keyboard]
  Port 4: 0103 power enable connect [1532:0067 Razer Razer Naga Trinity 00000000001A]
Current status for hub 1 [1d6b:0002 Linux 5.4.79-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]

When I run the command sudo uhubctl -l 1-1 -p 1 -a off -r 1000 it does not turn of the light and neither does it when I also turn of port 1 on hub number 2. But when I turn off the port where my mouse is connected to the mouse does turns off and stays off. Do you have any idea what might by the problem? I would greatly appreciate the help.

mvp, 2021-01-19:
Please follow FAQ for raspberry pi 4B: upgrade firmware, use uhubctl 2.3.0 or later, and you must control all 4 ports: `uhubctl -l 1-1 -a 0`. 1 port is not enough

mvp, 2021-01-21:
Closing. Feel free to reopen if this didn't answer your question - thanks!

closed 2021-01-21 by mvp

ChristiaanG, 2021-01-22:
Hi sorry for the late reply. My firmware version is 00138a1 and the version of uhubctl I'm using is 2.3.0. Do you mean with controlling all 4 points that I have to shut off power to all ports on hubs 1, 1-1 and 2? So it is not possible to shut off power to one specific port for this light by turning of power to that port on both the 1-1 and 2 hub?

mvp, 2021-01-22:
Hubs 1-1 and 2 are dual to each other, they represent USB 2 and USB 3 instances of the same physical USB 3 hub. Uhubctl should automatically turn off port for both dual hubs. So you can target either hub 1-1 or 2, it will yield the same result.
And yes, you have to turn off power to all 4 ports to cut power. Readme lists suggested command lines necessary for each Raspberry Pi model, including RPi4B: `uhubctl -l 2 -a 0` - not specifying -p means turn off for all ports of that hub.