mvp/uhubctl #310: syslog flooded with warnings

65sc02, 2021-01-25:
Hello,
on a RPI every call to uhubctl (even without any parameters) produces an entry in syslog:

kernel: [ 2363.842025] WARN::dwc_otg_hcd_hub_control:3808: DWC OTG HCD - Unknown hub control request type or invalid typeReq: 8006h wIndex: 0h wValue: f00h

This overflows my syslog.

mvp, 2021-01-25:
1. Which model of Raspberry Pi?
2. If you use RPi4B, did you upgrade firmware?
3. Which version of OS and kernel?
4. Why are you calling it every 2 seconds?

That said, calling uhubctl without parameters only does the following: enumerate all usb devices, filter hubs, then send control message to each hub to get device description. Getting your logs flooded might mean kernel bug, or you calling it so often might stress kernel.

65sc02, 2021-01-25:
I tried with a RPI-1 (yes, old) and a RPI 2B. I did not use a RPI4B. OS was Buster, latest, in both cases. As kernel I used a non-standard kernel:
Linux blumenwiese 5.10.9+ #1396 Thu Jan 21 15:50:23 GMT 2021 armv6l GNU/Linux
since the current kernel has a bug that sometimes disables CPU1 and USB at boot time, see
https://github.com/Hexxeh/rpi-firmware/issues/232 

This is also the reason why I am using uhubctl in /etc/watchdog.d/test-power.csh to check for missing USB power. Basically, what I am doing is
/sbin/uhubctl | /usr/bin/grep Port | /usr/bin/grep off
to see, if power is switched OFF for a port, which should not be the case, but it *is* the case if s.th. went wrong during boot (kernel bug). The watchdog will then reboot the RPI.

It is not a matter of call frequency: this happens with every call.

mvp, 2021-01-25:
I would blame your non standard kernel. Perhaps you could reduce call frequency to reduce logspam?

mvp, 2021-01-26:
I could not find such message `"Unknown hub control request type or invalid typeReq"` in any known official kernel tree at https://github.com/torvalds/linux or https://github.com/raspberrypi/linux.
It looks like your kernel is too custom - seems like it uses old proprietary dwc_otg driver.
I recommend to use standard RPi approved kernels instead - no one else has reported any issues with them ever before.

At any rate, not much I can do from uhubctl standpoint - it is not doing anything abnormal except for calling standard USB APIs.

closed 2021-01-26 by mvp

65sc02, 2021-01-26:
OK, thank you for looking into this. If this is a message from the kernel, then I am sorry to have bothered you - I thought it came from your program. Thank you, this is already a valuable information.

Bye
Helmut