mvp/uhubctl #311: Targus PAUH212

locostudios, 2021-02-05:
Bought this hub in hope of it working with Virtualhere and an iLok dongle. Haven't tried the VBUS thing cause i dont have a fan, light or a suitable cable to charge my phone in my studio at the moment, but the iLok is working fine remotely, and apparently it is dependent on a power cycling port.  I guess its safe to say this is working, especially since you already have the one with the "U" behind in your list already.

locostudios, 2021-02-05:
I've tried a few other hubs that turned out incompatible, or at least not working with an iLok. I list them here, might be useful for others.

Deltaco SuperSpeed Hub UH-472 Rev4
ORICO A3H7-US-BK
Hama 00078492

mvp, 2021-02-05:
Thanks for reporting! Added a note in device list about it.

commit 9b73c8e3 (2021-02-05): Note that PAUH212 is also supported

Closes #311.
closed 2021-02-05 by mvp