mvp/uhubctl #315: pkg-config from env variable in Makefil

coconutruben, 2021-02-12:
For the Makefile, the fact that this is calling `pkg-config` directly is causing issues with some of our toolchains on chromium os. Could you add a `PKG_CONFIG ?= pkg-config` line so that the build-scripts can pass our wrapped pkg-config?

mvp, 2021-02-12:
Implemented in 92736a8.

closed 2021-02-12 by mvp

coconutruben, 2021-02-12:
thank you so much, that's so quick! could it be possible to release 2.3.1 with this fix (and should I file an issue for that?)

mvp, 2021-02-12:
I should release 2.4.0 tomorrow. There is toggle feature that is still unreleased.

coconutruben, 2021-02-12:
awesome, thank you!

mvp, 2021-02-14:
@coconutruben, uhubctl 2.4.0 has been released, and contains this change https://github.com/mvp/uhubctl/releases/tag/v2.4.0

commit 92736a86 (2021-02-12): Honor PKG_CONFIG variable if exists

Some build systems want to pass location for pkg-config.

Closes #315.