mvp/uhubctl #316: Single smart hub works but daisy-chained hub doesn't

uugear, 2021-02-13:
Good day,

First I would like to thank you for making this good software!

We are developing a 4-port PPPS supported USB3.0 hub for Raspberry Pi 4B, and we would like to use uhubctl on it. The hub uses VL817 chip and also uses AP2511 as power switch for each port. We made some tests on the prototype and confirm the single hub works pretty well: we can use uhubctl to control the power on/off at each port, so I assume the hardware implementation is OK.

However when we daisy-chained two hubs, the hub at the second level doesn't fully work with uhubctl. The second hub was connected to port 1 in the first hub (hub 2-2), who connected to the lower USB 3.0 port (near the Raspberry Pi 4B board). So I tried to use this command to turn off the power on port 4 in the second hub.

`sudo ./uhubctl -l 2-2.1 -a off -p 4 -r 100`

We have power indicator and USB activity indicator on every port, so I can tell the power at that port was not cut. We then connected a USB drive on port 4 in the second hub, and repeated the command above. This time I saw the USB activity indictor LED turned off, which means the data was cut, however the power was still not cut.

I then used this command to restore the port 4 in the second hub:

 `sudo ./uhubctl -l 2-2.1 -a on -p 4`

And the USB activity indicator LED turned on again, so the data connection was restored.

So it seems the single hub can work with uhubctl properly, while the daisy-chained can only turn on/off the data. Could you advice me a direction on troubleshooting this kind of issue? Thank you in advance!

mvp, 2021-02-13:
Relevant paragraph from FAQ:
> If you have USB 3.0 hub connected to USB3 upstream port, it will be detected as 2 independent virtual hubs: USB2 and USB3, and your USB devices will be connected to USB2 or USB3 virtual hub depending on their capabilities and connection speed. To control power for such hubs, it is necessary to turn off/on power on both USB2 and USB3 virtual hubs for power off/on changes to take effect. uhubctl will try to do this automatically (unless you disable this behavior with option -e).

Keyword here is `uhubctl will try to do this automatically`. If it fails to find proper USB2/USB3 hub pair, power will still remain on on physical USB port.
I suggest to see which ports uhubctl is trying to turn off, and make sure they are correct hub pair.
For example, this is one of my USB3 hubs. Notice that uhubctl automatically detected it has USB2/USB3 pair:
```
$ uhubctl -l 3-3
Current status for hub 4-3 [174c:3074 Asmedia ASM107x 00000000A001, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 3-3 [174c:2074 Asmedia ASM107x 00000000A001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0303 power lowspeed enable connect [0e9c:0000 Streamzap, Inc. Streamzap Remote Control]
```
When I try to turn off port 4, it turns off both USB2 and USB3 ports:
```
$ uhubctl -l 3-3 -p 4 -a 0
Current status for hub 4-3 [174c:3074 Asmedia ASM107x 00000000A001, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 4-3 [174c:3074 Asmedia ASM107x 00000000A001, USB 3.00, 4 ports, ppps]
  Port 4: 00a0 off
Current status for hub 3-3 [174c:2074 Asmedia ASM107x 00000000A001, USB 2.10, 4 ports, ppps]
  Port 4: 0303 power lowspeed enable connect [0e9c:0000 Streamzap, Inc. Streamzap Remote Control]
Sent power off request
New status for hub 3-3 [174c:2074 Asmedia ASM107x 00000000A001, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
```
and indeed status shows power is off on port 4 of both hubs:
```
$ uhubctl -l 3-3
Current status for hub 4-3 [174c:3074 Asmedia ASM107x 00000000A001, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 00a0 off
Current status for hub 3-3 [174c:2074 Asmedia ASM107x 00000000A001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0000 off
```
If I turn on power on either USB2 or USB3 hub with `-e` (ignoring duality handling), it will keep other one off, but will immediately restore vbus to physical port:
```
$ uhubctl -l 3-3 -p 4 -a 1 -e
Current status for hub 3-3 [174c:2074 Asmedia ASM107x 00000000A001, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
Sent power on request
New status for hub 3-3 [174c:2074 Asmedia ASM107x 00000000A001, USB 2.10, 4 ports, ppps]
  Port 4: 0100 power
```
status:
```
$ uhubctl -l 3-3
Current status for hub 4-3 [174c:3074 Asmedia ASM107x 00000000A001, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 00a0 off
Current status for hub 3-3 [174c:2074 Asmedia ASM107x 00000000A001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0303 power lowspeed enable connect [0e9c:0000 Streamzap, Inc. Streamzap Remote Control]
```

Basically, my advice is to ensure that both USB2 and USB3 ports are turned off. In case uhubctl gets confused and cannot identify pair, use `-e` to fix it manually on corresponding hub.

And if you are sure to turn off both and still see VBUS, your hardware is not compliant and needs to be fixed.

closed 2021-02-13 by mvp

SlySven, 2021-02-13:
> And if you are sure to turn off both and still see VBUS, your hardware is not compliant and needs to be fixed.

@mvp please note what the OP was saying:
> ... We are developing a 4-port PPPS supported USB3.0 hub for Raspberry Pi 4B, and we would like to use uhubctl on it. The hub uses VL817 chip and also uses AP2511 as power switch for each port....

They want to produce **real** hardware that works properly and thus is fully in accordance with the requirements - given how small the proportion of Hubs on the market that do that are, RPi end-users, others and **uhubctrl** should be encouraging and supporting them (https://www.uugear.com/) to *get it right*!

mvp, 2021-02-13:
@uugear, I didn't mean to discourage you :-)
I just explained what could be wrong. @SlySven is right!
If your hub does turn off VBUS when connected to Linux or Mac host (connected phone stops charging), then your hardware must be compliant.
If it doesn't work when connected via RPI4B, it is most likely an issue with uhubctl not finding proper hub pair and turning off power on both USB 2 and USB 3 components of your hub.

uugear, 2021-02-14:
@mvp thank you for the advice!

When we connect only one hub to Raspberry Pi 4B, we can use uhubctl to properly turn off VBUS for each port. We have an LED indicates the VBUS for each port, so we can confirm that VBUS can be turned off.

I checked it again and it seems uhubctl can find the USB2/USB3 hub pair correctly. Here is the output of uhubctl without parameter:
```
pi@raspberrypi:~/uhubctl $ sudo ./uhubctl 
Current status for hub 2-2.1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2 [1d6b:0003 Linux 5.4.51-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.2.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.2 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 5.4.51-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
```
Hub 2-2 is the first hub I connected to Raspberry Pi 4B, and it has its paired USB 2.0 hub "hub 1-1.2", while hub 2-2.1 is connected to the port 1 on hub 2-2, and it has its paired USB 2.0 hub "hub 1-1.2.1". Below is the output when I run "sudo ./uhubctl -l 2-2.1 -a off -p 4 -r 100" command:
```
Current status for hub 2-2.1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-2.1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.2.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0100 power
Sent power off request
New status for hub 1-1.2.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
```
I can see uhubctl did sent power off requests to hub 2-2.1 and hub 1-1.2.1. However the port 4 on hub 2-2.1 got "02a0 power 5gbps Rx.Detect" as new status, while the port 4 on hub 1-1.2.1 got "0000 off" as new status. I confirmed the port status again by running "sudo ./uhubctl":
```
pi@raspberrypi:~/uhubctl $ sudo ./uhubctl 
Current status for hub 2-2.1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2 [1d6b:0003 Linux 5.4.51-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.2.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0000 off
Current status for hub 1-1.2 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 5.4.51-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
```
So indeed only hub 1-1.2.1 changed its port 4 status to off, and it seems the power off request to hub 2-2.1 was ignored. Do you have idea what could cause this kind of problem?

mvp, 2021-02-14:
Could you please try turning off power for your hub separately on USB 2 and USB 3, that is, issue these 2 commands;
```
uhubctl -l 1-1.2.1 -p 4 -a 0 -e
uhubctl -l 2-2.1 -p 4 -a 0 -e
```
Then verify that power turns off. If it doesn't, try repeating these commands, perhaps add `-r 1000` too.
Also check if it depends on order - your hub may require turning off USB 2 and USB 3 in certain order.
Normally, you should not use `-r` option though, it is only useful if there was device connected on target port, and Linux kernel gets surprised by device disappearing, this trying to re-power it back on. In such case -r could help to suppress that. I don't think you have such situation here though.
At any rate, as soon as both USB 2 and USB 3 ports are off, VBUS should go down.

uugear, 2021-02-14:
We found that the -r option is a must in our case, without "-r 100" the hub could not turn off the port with USB device connected to it (power always comes back after a few seconds), and that happens even on the hub directly connected to Raspberry Pi 4B. 

I ran command "uhubctl -l 1-1.2.1 -p 4 -a 0 -e", and got:
```
pi@raspberrypi:~/uhubctl $ sudo ./uhubctl -l 1-1.2.1 -p 4 -a 0 -e
Current status for hub 1-1.2.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0100 power
Sent power off request
New status for hub 1-1.2.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
```
As expected, power stayed at port 4. Then I ran "uhubctl -l 2-2.1 -p 4 -a 0 -e", and got:
```
pi@raspberrypi:~/uhubctl $ sudo ./uhubctl -l 2-2.1 -p 4 -a 0 -e
Current status for hub 2-2.1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-2.1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
```
Still the power at port 4 was not cut. 

I also tried to reverse the order of commands, and then tried adding "-r 1000" to each of them, the results were the same: the daisy-chained hub 2-2.1 seems always ignore the power off request, and its port 4 always have the status "02a0 power 5gbps Rx.Detect".

I wonder if there is anything else we could try?

I checked the list of known compatible USB hubs, and found these three hubs use the same VL817 chip, and they have 2 or 3 hubs internally daisy-chained:
AmazonBasics | HU9003V1EBL | 7 | 3.1 | 2109:2817 | 2018 |  
AmazonBasics | HU9002V1SBL, HU9002V1ESL | 10 | 3.1 | 2109:2817 | 2018 |  
AmazonBasics | HUC9002V1SBL, HUC9002V1EBL, HUC9002V1ESL | 10 | 3.1 | 2109:2817 | 2018

According to [issue 261](https://github.com/mvp/uhubctl/issues/261), the daisy-chained VL817 hub should also work, as long as the hub pair is correctly specified. I have no idea what we did wrong, and would like to know your thoughts. Thank you.

mvp, 2021-02-14:
I don't know why is that. uhubctl is definitely sending proper control message to the hub, and then it is reading current port state back again. If hub does not turn power off, it must be VL817 chip issue.

uugear, 2021-02-15:
@mvp thank you for the insight, and I think you are right.

It seems to be firmware related. AmazonBasics' working hubs are all using firmware revision 90.23 for VL817, while we are using firmware revision 90.11.

We also tried the default firmware 00.50 and revision 90.33 (downloaded from Internet), they behaviour are slightly different, but unfortunately none of them can fully work.

We have contacted VIA-Labs and asked if they could provide us the firmware revision 90.23. This could take some time, and we don't even know if we will eventually have it.

mvp, 2021-02-15:
Good luck with getting Via support. I remember that Raspberry Pi foundation had similar issue with VL805 chipset for RPi4B, and Via eventually delivered working firmware.

As for needing -r 1000 because power comes back, that is not necessarily hardware problem. Most likely this is Linux USB kernel driver issue as mentioned in FAQ - basically, even though kernel knows that device power is off, it doesn't propagate this knowledge properly through the stack (including to udev), and drivers are confused by disappearing devices trying to bring power back to them. This is not an issue on other operating systems like MacOS or *BSD. Adding -r 1000 keeps power off long enough that most Linux drivers give up trying to revive disappeared device.

uugear, 2021-02-17:
Thank you @mvp for the information.

I understand the reason to need the -r parameter, and I think it is totally acceptable.

Do you think it is possible to dump the firmware file from a working hub, so we can test it on our own hubs?

mvp, 2021-02-17:
@uugear, I don't know if it is possible to extract firmware from other hubs. It is completely dependent on firmware utilities provided by chipset manufacturer like Via, Microchip, TI, Realtek or others.

uugear, 2021-08-09:
@mvp we managed to obtain the 90.23 firmware from Via Labs, and it indeed works well. Our product has been released: 
[MEGA4 product announcement](https://www.uugear.com/news/new-product-mega4-usb-3-1-ppps-hub-for-raspberry-pi-4/)
[MEGA4 product page on e-shop](https://www.uugear.com/product/mega4-4-port-usb-3-ppps-hub-for-raspberry-pi-4b/)

@mvp I would like to thank you for the help and want to send you a free MEGA4 hub board. Please drop me an email (to shawn at uugear dot com) and let me know your shipping address and telephone number (for UPS delivery only). Thank you.

We also have gifts to all people who are interested in this project. Please use the coupon code "mega4_half_off" during the checkout and you get 50% off for your MEGA4 hub board (apply on one unit only). This coupon code will expire on August 15th 2021. Please feel free to share this coupon code to whom might be interested :)

mvp, 2021-08-09:
Thank you for hard work @uugear , I have added MEGA4 to supported device list.

mvp, 2021-08-12:
@uugear, I have received your MEGA4 sample. I can confirm it works perfectly - tried it on Mac and Linux (not on RPI4B yet).
I love LED lights showing port status too - this makes it one of the best PPPS hubs out there, and at affordable place. Much thanks!

mvp, 2021-08-12:
Added a note about MEGA4 in README in bc80baa

uugear, 2021-08-12:
@mvp I am very glad that you like it :)

SlySven, 2021-08-22:
@uugear Thanks also for the sample you sent me which I picked up a couple of days ago! It works as expected - even though i am only using it in USB 2 mode. Have you considered offering an offset inter-PWB USB connector for the older RPi models - I can't recall whether the RPi 3's do USB 3 or not.

\<aside>As another RPi project have you considered a UPS type power supply unit that can connect to an external battery (a Lead-Acid for cheapness or a Lithium cell/pack for power density) and/or a solar/wind generator as an external source?

I own a *Basic* model "UPiS" made by PiModules - which does some of this see https://www.raspberrypi.org/forums/viewtopic.php?f=37&t=44989 or the abandoned forums http://108.175.10.239/index.php  - the manufacturer of it has moved on to shinier products and their current web-site doesn't even list it any more {and the 2.0 firmware revision promised many years ago never materialised}\</aside>

uugear, 2021-08-22:
@SlySven we already considered that, and we most probably will do so in the future. However only RPi 4 has USB 3 ports, so other RPi models can only use MEGA4 as USB 2 hub.

As for power supply, we have a product named [Zero2Go Omini](https://www.uugear.com/product/zero2go-omini-wide-input-range-multi-channel-power-supply-for-raspberry-pi/), which is very close to what you described. It has three input channels so you can connect battery and solar panel at the same time. You can use it to implement a UPS, and you can find an example on page 18 in [its user manual](https://www.uugear.com/doc/Zero2Go_Omini_Rev2_UserManual.pdf). However you will need to add your own charging circuit if you use battery.

damonfloyd, 2022-02-28:
@uugear In the event we are not as successful in obtaining the 90.23 firmware from via-labs, can you upload or post a link to it?