mvp/uhubctl #317: D-Link Revision number unclear

pm-pm, 2021-02-14:
Hello.
I have a question: What das "B7+" mean?

From your site:
"DUB-H4 rev B (silver). Note: rev B7+ not supported"

Is "B7+" a standalone revision?
Is there a difference between "B7" and "B7+"?
Or does it mean "B7" and later like "B7", "B8", "B9", ... ?

Here you find "B7E with HW B7" and i am confused.
https://www.ebay.de/itm/USB-Hub-D-LINK-DUB-H4-HW-B7-4fach-USB-Hub-kein-Netzteil-kein-Kabel/124573837418

mvp, 2021-02-14:
B7+ means rev B7 and possibly anything later. We only had report with B7 not working though. If you find that B7E works, please let us know!

pm-pm, 2021-02-14:
Thanks for your response. 
I do not own the "B7E" Hub. It was the question whether i should buy it?
No. ;-)

But i have success with the two USB 3.0 Port of "[HPE ProLiant MicroServer Gen8](https://support.hpe.com/hpesc/public/docDisplay?docId=c03793483)" :-)
The USB 2.0 ports do not work.

I will do a more detailed info later. ;-)

mvp, 2021-02-14:
Thank you! Due to multitude of all USB hubs in the wild, I cannot possibly test all of them, so I have to rely on user reports like yours :-).
My question was if you happen to run into any hub that definitely works but not listed, or one that is listed but doesn't work, please report it!

pm-pm, 2021-02-14:
This is how I will do it. :-)
Thanks for your work and support!

Note:  "Backup Plus Hub STEL8000100"
You need the external 12V Power to operate in "hub only" mode. (It was expactable)
It is NOT possible to use bus power for PCB after removing the HDD. You need the external 12V.

mvp, 2021-02-15:
Thanks for info. Closing this

closed 2021-02-15 by mvp