mvp/uhubctl #319: TP-Link USB 3.0 7-Port HUB

mkettn, 2021-02-24:
Hi,
I have a 7-Port USB Hub from TP-Link. According to lsusb it has per port power switching. 
I can see active USB ports on the hub with a small white LED, but if I turn off the port using uhubctl, the device is still powered. It looks like just the data wires are disconnected and the white LED on the port is turned off. For example a connected Arduino still is powered on, but I cannot program it or use the serial connection. Am I dumb or does the USB Hub don't support uhubctl?
To turn off the port I used this command: `sudo uhubctl -a off -l 3-9.2 -p 4`

mvp, 2021-02-24:
There is small possiblity that turning all ports off would cut off VBUS. But in general you are probably right that manufacturer saved few cents on adding proper circuitry to actually cut off VBUS per port.

mvp, 2021-02-25:
Feel free to reopen if you still have questions

closed 2021-02-25 by mvp