mvp/uhubctl #320: UE-1008 8-Port USB 3.0 PCIe Expansion Card

seebq, 2021-02-26:
Great tool, thanks for creating this.  We're having trouble getting this tool to work with the PCIe expansion card based on the "uPD720202" chipset to work:

`vendor: 1912 ("Renesas Technology Corp."), device: 0015 ("uPD720202 USB 3.0 Host Controller")`

And wondering if it has to do with the USB 3.0 duality setup.  Here's what the tool reports on our PC.  

Note, this system also has some internal USB hubs (GenesysLogic hubs), displaying entire output here, censoring a few devices, but leaving our RealSense cameras plugged into the PCIe expansion card, for fun:

```
sudo ./uhubctl
Current status for hub 2-2.4 [05e3:0620 GenesysLogic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2 [05e3:0620 GenesysLogic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0263 power 5gbps U3 enable connect [05e3:0620 GenesysLogic USB3.2 Hub, USB 3.20, 4 ports, ppps]
Current status for hub 1-2.4 [05e3:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-2 [05e3:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0103 power enable connect [06xx:00xx <censored device here> ]
  Port 2: 0103 power enable connect [15xx:0xax <censored device here> ]
  Port 3: 0100 power
  Port 4: 0507 power highspeed suspend enable connect [05e3:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports, ppps]
Current status for hub 9 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:06:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0107 power suspend enable connect [04xx:60xx <censored device here> ]
  Port 2: 0100 power
Current status for hub 10 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:06:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 8 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:05:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051361]
  Port 2: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051747]
Current status for hub 7 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:05:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 6 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:04:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051475]
  Port 2: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051436]
Current status for hub 5 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:04:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 4 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 025523050879]
  Port 2: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223050372]
Current status for hub 3 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
```

Note the two hubs each for this 8 port device, one for the USB 3.00 and USB 2.00 ports.  The device won't let you cycle the power on the USB 3 hub location, i.e. if you try to power off (or power on or cycle) on hub 4, it complains:

```
sudo ./uhubctl -l 4 -p 1 -a off
No compatible devices detected at location 4!
Run with -h to get usage info.
```

You can do all actions on the USB 2.00 hub device, though:
```
sudo ./uhubctl -l 3 -p 1 -a on
Current status for hub 3 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0000 off
Sent power on request
New status for hub 3 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
```

Is there an issue with the automatic USB 3.0/2.0 detection because the hubs have completely different sequential numbers?  Note that we can reset it using the `-e` option to force on the 3.00 hub, but our camera device actually ends up jumping over to the 2.00 port quickly.  Would love to do both at the same time or use the duality handling code.

Is there a better way of identifying these devices or manually or automatically handling the USB3.0/2.0 duality nature of this hub?

I'd love to update the documentation with more examples of matching on search and device strings once we get this working.

Thanks again.

mvp, 2021-02-26:
Duality matching code significantly depends on devices advertising ContainerID BOS (binary object store) descriptor, which per USB3 spec should be globally unique but reported the same whether device operates in USB2 or USB3 mode.
Unfortunately, Linux kernel does not advertise ContainerID for root USB3 hubs - and your hub 4 is root USB3 hub.
This makes almost impossible to reliably match correct USB2 hub for that root USB3 hub.
From here, we have few options:

* Beg Linux kernel USB developers to advertise ContainerID for root hubs - ideal, but we have no idea when it could possibly happen)
* Use `-e` workaround to match them manually - kludgy, but you can do it now without having to wait for anyone
* Try to add even more hacks into duality matching code, trying to match even if ContainerID is missing - not impossible, but rather difficult, and highly unreliable.

Note that I actually added hack for this very issue that makes Raspberry 4B work - they also have root USB3 hub which does not advertise container id. Theoretically, it is possible to keep adding hacks like that, but I would rather not make this rathole even deeper.

seebq, 2021-02-26:
Thanks for the quick response on this, and understood.

So by using the -e though, they drop off entirely and are gone.  Is there a desired order to do the -e or both at same time?

```
sudo ./uhubctl
<snip>
Current status for hub 9 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:06:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0103 power enable connect [0403:6001 FTDI USB-RS232 Cable FT40AMRP]
  Port 2: 0100 power
Current status for hub 10 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:06:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 8 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:05:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051361]
  Port 2: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051747]
Current status for hub 7 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:05:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 6 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:04:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051475]
  Port 2: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051436]
Current status for hub 5 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:04:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 4 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 025523050879]
  Port 2: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223050372]
Current status for hub 3 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power

# turn off the USB 2.00 hub port first:
sudo ./uhubctl --action off --location 3 --ports 1 --exact
Current status for hub 3 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
Sent power off request
New status for hub 3 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0000 off

sudo ./uhubctl
<snip>
Current status for hub 9 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:06:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0103 power enable connect [0403:6001 FTDI USB-RS232 Cable FT40AMRP]
  Port 2: 0100 power
Current status for hub 10 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:06:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 8 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:05:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051361]
  Port 2: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051747]
Current status for hub 7 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:05:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 6 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:04:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051475]
  Port 2: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223051436]
Current status for hub 5 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:04:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 4 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 3.00, 2 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 025523050879]
  Port 2: 0203 power 5gbps U0 enable connect [8086:0b5c Intel(R) RealSense(TM) Depth Camera 455 Intel(R) RealSense(TM) Depth Camera 455 028223050372]
Current status for hub 3 [1d6b:0002 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 2.00, 2 ports, ppps]
  Port 1: 0000 off
  Port 2: 0100 power

# now turn off the USB 3.00 hub port next:
sudo ./uhubctl --action off --location 4 --ports 1 --exact
Current status for hub 4 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:03:00.0, USB 3.00, 2 ports, ppps]
[... 18 lines cut ...]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 6 [1d6b:0003 Linux 5.10.8nlb xhci-hcd xHCI Host Controller 0000:04:00.0, USB 3.00, 2 ports, ppps]
[... cut ...]
[   93.366184] xhci_hcd 0000:03:00.0: xHCI host controller not responding, assume dead
[   93.366200] xhci_hcd 0000:03:00.0: HC died; cleaning up
[   93.366215] usb 4-1: USB disconnect, device number 2
```

Any thoughts on how to do this with -e, or do both of them together with a --vendor or --search instead?

And as for the hacks on duality matching, understood, I agree.  I am curious if I could hardcode it for this one controller for my own purposes, but realize it might not go back upstream.

Thanks again for the quick response.  Great tool.

mvp, 2021-02-26:
Order of turning off USB2 vs USB3 should not really matter.
However, note that if your target device is USB3, and you turn off USB3 port first, it should drop but quickly reconnect as USB2 device.
After you turn off USB2 port, it will be gone for good.
To minimize that disturbance and your target device reconnecting, you should turn off first whichever bus it is not actively connected to first.

Regarding searching, note that hub locations (aka address) are semi-stable - locations should never change as long as you keep USB topology the same, e.g. connect/disconnect USB device tree into the same physical ports. So, you should be able to put them into your configs without fear of them changing randomly.

mvp, 2021-02-26:
Sorry missed that point about hub completely disappearing.
In general, root hubs are rarely supporting power switching properly - because they are soldered on motherboard itself and typically can rely on VBUS always present from main power supply.

I guess in your case, turning it off exposed some hardware bug or issue so they completely gone from USB busses. Not sure if this is something that can be fixed in Linux kernel, or actual hardware problem.

As a workaround, I recommend to obtain any of the hubs in supported table (and there are plenty of USB3 hubs there), attach to your host and control ports as you wish on that external hub instead.

mvp, 2021-02-28:
I don't see this as anything that can be fixed or improved by uhubctl. Please reopen this if you have any more details or think this can be fixed by uhubctl.

closed 2021-02-28 by mvp

seebq, 2021-03-01:
Meant to respond and say thanks, and should have closed it myself.  Thx again!