mvp/uhubctl #321: ASMedia AS2107: Strange behaviour

luke-jr, 2021-02-28:
"off" does nothing, while "on" turns the power off (only way to get it back is to unplug and plug back in)

mvp, 2021-02-28:
Can you please provide more information?

luke-jr, 2021-02-28:
Such as?

Device is a Pixel XL if that matters

mvp, 2021-02-28:
Your end device makes no difference - USB hub does.
Please provide output for `sudo uhubctl`, also what commands you are executing, what result you are observing, what works and what doesn't. Right now, it impossible to diagnose your problem.

luke-jr, 2021-02-28:
```
Current status for hub 3-1 [8087:8000, USB 2.00, 8 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
  Port 8: 0100 power
Current status for hub 1-1 [8087:8008, USB 2.00, 6 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
Current status for hub 4-5 [174c:3074 ASMedia AS2107 USB2.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0203 power 5gbps U0 enable connect [18d1:4ee7 Google Pixel XL HT74W0201650]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-9 [174c:2074 ASMedia AS2107 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

`uhubctl -l 4-5 -p 3 -a on` -> nothing
`uhubctl -l 4-5 -p 3 -a off` -> nothing
`uhubctl -l 4-5 -p 3 -a on` -> Pixel XL shows charging stop (but Ampere suggests current is still flowing??)
`uhubctl -l 4-5 -p 3 -a on` -> nothing
`uhubctl -l 4-5 -p 3 -a off` -> nothing

mvp, 2021-02-28:
Please clarify what nothing means here:

> uhubctl -l 4-5 -p 3 -a on -> nothing
uhubctl -l 4-5 -p 3 -a off -> nothing

Ideally, just show output of running those commands.

If port is `off` according to uhubctl output but connected device is still charging or otherwise gets power (e.g. according to ampermeter), it just means that your USB hub does not support power switching properly.

Note that only USB hubs listed in supported table are (more or less) guaranteed to work for power switching. Your random Asmedia hub may or may not support it.

luke-jr, 2021-02-28:
```
# uhubctl -l 4-5 -p 3 -a on
Current status for hub 4-5 [174c:3074 ASMedia AS2107 USB2.0 Hub, USB 3.00, 4 ports, ppps]
  Port 3: 0203 power 5gbps U0 enable connect [18d1:4ee7 Google Pixel XL HT74W0201650]
Sent power on request
New status for hub 4-5 [174c:3074 ASMedia AS2107 USB2.0 Hub, USB 3.00, 4 ports, ppps]
  Port 3: 0203 power 5gbps U0 enable connect [18d1:4ee7 Google Pixel XL HT74W0201650]
Current status for hub 2-9 [174c:2074 ASMedia AS2107 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0100 power
Sent power on request
New status for hub 2-9 [174c:2074 ASMedia AS2107 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0100 power
# uhubctl -l 4-5 -p 3 -a off
Current status for hub 4-5 [174c:3074 ASMedia AS2107 USB2.0 Hub, USB 3.00, 4 ports, ppps]
  Port 3: 0203 power 5gbps U0 enable connect [18d1:4ee7 Google Pixel XL HT74W0201650]
Sent power off request
New status for hub 4-5 [174c:3074 ASMedia AS2107 USB2.0 Hub, USB 3.00, 4 ports, ppps]
  Port 3: 00a0 off
Current status for hub 2-9 [174c:2074 ASMedia AS2107 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0100 power
Sent power off request
New status for hub 2-9 [174c:2074 ASMedia AS2107 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0000 off
# uhubctl -l 4-5 -p 3 -a on
Current status for hub 4-5 [174c:3074 ASMedia AS2107 USB2.0 Hub, USB 3.00, 4 ports, ppps]
  Port 3: 00a0 off
Sent power on request
New status for hub 4-5 [174c:3074 ASMedia AS2107 USB2.0 Hub, USB 3.00, 4 ports, ppps]
  Port 3: 00a0 off
Current status for hub 2-9 [174c:2074 ASMedia AS2107 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0000 off
Sent power on request
New status for hub 2-9 [174c:2074 ASMedia AS2107 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0100 power
```

The "hub" is built-in to my ASRock Z87 Extreme4 motherboard.

mvp, 2021-02-28:
Again, only USB hubs listed in supported table are known to work for power switching.
Your motherboard's built-in USB hub may or may not support it.
Looking at your output, I don't see anything abnormal - uhubctl is doing what it is supposed to do, it is able to properly correlate dual USB2/USB3 hubs, and power state reported looks consistent.
If you turn power off, but voltage is still present, it simply means that your USB hub does not properly support power switching.
If your built-in USB hubs do not support VBUS control, the only workaround is to obtain any hub from supported table, connect your devices via such hub and control power on it instead.

mvp, 2021-02-28:
Please reopen if you have more details or believe this is software bug.

closed 2021-02-28 by mvp

luke-jr, 2021-03-02:
Just an update: This appears to have completely messed up my motherboard's USB stack. Hard drives on different internal USB hubs started dropping off even long after I had ceased playing with uhubctl, and even reloading the kernel modules didn't help for more than a few minutes. So something definitely very wrong with the hardware, sadly. :(

mvp, 2021-03-02:
And even reboot doesn't fix it?

luke-jr, 2021-03-02:
I didn't try a warm reboot, but cold does seem to have fixed it (time will tell for sure).

mvp, 2021-03-02:
Good to know that you got your computer working. I never heard of USB hardware being broken by uhubctl. And I have used it to turn power on/off on many various hubs many times a day for years.