mvp/uhubctl #323: Power on not working on Port

acocalypso, 2021-03-08:
I've been using uhubctl for quite a while but since 3 hours I have a problem with on port that is no longer able to start.

The port stays off even after I execute the power on command.

```
pi@raspberrypi:~ $ sudo uhubctl -a on -p 3 -l 1-1.2.4
Current status for hub 1-1.2.4 [2109:2817 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0000 off
Sent power on request
New status for hub 1-1.2.4 [2109:2817 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0000 off

```

I'm using a supported amazon basic 10 Port hub, is there a possibility that the usb port get's damaged after a few off / on?

closed 2021-03-08 by acocalypso

mvp, 2021-03-09:
I see that you closed this quickly - did you figure out what was wrong?

Regarding possible damage - I had many hubs (not AmazonBasics though) to switch power on/off 5-100 times a day for many end devices for years, problems were really rare (perhaps 1 hub out of about 40 has died in 3 years). But remember that any electronics could go bad after long use, e.g. typical reason is capacitors drying out.

acocalypso, 2021-03-09:
A simple removal of power to the hub solved the problem for me…

Thanks for information about possible damages.