mvp/uhubctl #326: Raspberry 4B, USB turns back on ms later

Alexwijn, 2021-03-21:
I'm using a USB powered led strip that I to control with uhubctl, but it stopped working.

I just upgraded from a Pi3 to a Pi4 (while still using the same SD card). And the command that I always used stopped working:
`sudo uhubctl -l 1-1 -a 0`

So I changed the command according the README, to turn off the USB2 ports:
`sudo uhubctl -l 2 -a 0`

But once I executed the above command the USB2 port will power off but powered itself back on within a second. I also tried to set various `authorized` files to `0` without any luck.

Any help is much apricated, thanks!

mvp, 2021-03-21:
Make sure you have upgraded firmware per readme. If power comes back on, that is also covered on FAQ, try adding -r 1000

Alexwijn, 2021-03-21:
Thank you for the quick response.

I'm running on the firmware from September 2020 (000138a1), no updates are available.
I also tried the `-r` before without any luck. Could you specify what else I could try?

mvp, 2021-03-21:
Do you have any other usb devices connected to your Pi, except for led strip? Try disconnecting anything except for led strip. If that helps, proper solution is to set authorized=0 for devices which have their kernel device drivers to cause reconnection. Or, use use even higher value for -r

Alexwijn, 2021-03-21:
Yes, I had one other device connected to the USB3. Tried it without but that did not change anything.

I'm not sure if it was caused by setting the `authorized=0` or removing the second device, but the blinking stops and it won't turn off at all now.

mvp, 2021-03-21:
That's weird. Few questions:
* Are you using uhubctl 2.4.0?
* Which command you use to turn power off? I recommend `sudo uhubctl -l 2 -a 0` or `sudo uhubctl -l 1-1 -a 0` - that turns power off for all 4 ports
* If you use it without sudo, make sure you set udev permissions for both USB 2 and USB 3 chips (per readme).
* please post output of running `uhubctl -l 2 -a 0`

Alexwijn, 2021-03-21:
No, I'm using 2.0.0-5. I installed it via `apt-get`. Any reason why the repo is out-of-date?

```
> sudo uhubctl -l 2 -a 0
Current status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Sent power off request
New status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1 [1d6b:0002 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
Sent power off request
New status for hub 1 [1d6b:0002 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports]
  Port 1: 0000 off
```

mvp, 2021-03-21:
You need at least version 2.3.0 for this to work on RPi4B. Regarding obsolete repositories, that's a question for raspbian maintainers.

Alexwijn, 2021-03-21:
Do you know where I can find an alternative repo or a pre-compiled .dep file? Or should I just compile it?

mvp, 2021-03-21:
just compile it:

    git clone https://github.com/mvp/uhubctl
    cd uhubctl
    sudo make install

Alexwijn, 2021-03-21:
Thanks, also don't forget to compile it first:
```
git clone https://github.com/mvp/uhubctl
cd uhubctl
make
sudo make install
```

It is working now with the latest version, thanks again! Maybe add a note to the README that the raspbian repo is out-of-date?

mvp, 2021-03-21:
Glad you got it working :-)
I will add a note to not file bugs against old uhubctl versions instead.

closed 2021-03-21 by mvp