mvp/uhubctl #328: D-Link DUB-H4 H/W ver: B4 silver does NOT power off ports

khain0, 2021-03-28:
Description: USB ports are not switched off (stay powered on)
Hardware: Silver D-Link DUB-H4 H/W ver: B4 connected to Raspberry Pi 3 Model B Rev 1.2
ID 05e3:0606 Genesys Logic, Inc. USB 2.0 Hub / D-Link DUB-H4 USB 2.0 Hub
Software: raspbian buster Linux rpi3 5.4.83-v7+ and uhubctl 2.3.0-7-g6ac502a0
Steps to reproduce:
1) connect hub to raspberry using USB cabel
2) perform command ad root: uhubctl -l 1-1.4 -a off -p all

Expected result: all ports stays powered off
Actual result: uhubctl indicates that ports are powered off, but USB led flashes after connecting to one of hub ports
root@rpi3:~# uhubctl -l 1-1.4 -a off -p all
Current status for hub 1-1.4 [05e3:0606 USB2.0 Hub, USB 2.00, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Sent power off request
New status for hub 1-1.4 [05e3:0606 USB2.0 Hub, USB 2.00, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off

Characterization isolation:
root@rpi3:~# lsusb -v 2>/dev/null |grep -A76 "05e3:0606"
Bus 001 Device 010: ID 05e3:0606 Genesys Logic, Inc. USB 2.0 Hub / D-Link DUB-H4 USB 2.0 Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0606 USB 2.0 Hub / D-Link DUB-H4 USB 2.0 Hub
  bcdDevice            7.02
  iManufacturer           0
  iProduct                1 USB2.0 Hub
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x00e9
    Per-port power switching
    Per-port overcurrent protection
    TT think time 32 FS bits
    Port indicators
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent    100 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0000
   Port 2: 0000.0000
   Port 3: 0000.0000
   Port 4: 0000.0000
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         0 Full speed (or root) hub
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
  Self Powered

mvp, 2021-03-28:
LED is not reliable indicator for port powered on or off. Can you please check if connected phone stops charging? Or USB fan stops spinning? Or just check voltage. Thanks

mvp, 2021-03-28:
If you observe that power is not turning off, I will change readme to mention that rev B4+ are not supported...

closed 2021-03-29 by mvp

khain0, 2021-03-30:
I meant USB LED connected to USB port still flashes, not the led indicator on hub.
5V is still supplied after powering off USB port using uhubctl. I have checked using multimeter.
The same behaviour is using hub-ctrl command.
Seems device with ID 05e3:0606 does not support powering off USB ports even if lsusb outputs "Per-port power switching"

mvp, 2021-03-31:
Thanks, I updated readme to mention that B4 does not work

commit 2820f9fa (2021-03-29): DLink DUB-H4 rev B4 does not work

Closes #328.