mvp/uhubctl #330: Power not being cut from device with AmazonBasics HU3641V1

GravesJake, 2021-03-29:
Running on Ubuntu 20.04 with the AmazonBasics HU3641V1. It's connected to my laptop with a usb-c adapter because there are only usb-c ports on my system. 

I get I/O errors when trying to run my device-info command after turning the port off but the LED stays on and power isn't actually cut from the device.
```
$ sudo uhubctl
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [1d50:60e6 TARGET DEVICE *************************]
  Port 2: 0100 power
  Port 3: 0103 power enable connect [046d:c539 Logitech USB Receiver]
  Port 4: 0103 power enable connect [04d9:0296 USB-HID Keyboard]
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0489:e0a2 Foxconn / Hon Hai
Bus 001 Device 003: ID 0bda:58f4 Realtek Semiconductor Corp. Integrated_Webcam_HD
Bus 001 Device 007: ID 27c6:5385 Shenzhen Goodix Technology Co.,Ltd. Fingerprint Reader
Bus 001 Device 008: ID 04d9:0296 Holtek Semiconductor, Inc.
Bus 001 Device 006: ID 046d:c539 Logitech, Inc.
Bus 001 Device 010: ID 1d50:60e6 TARGET DEVICE ******************************************
Bus 001 Device 002: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
$ device-info
  Found TARGET DEVICE *************************
  firmware info...

$ sudo uhubctl -p 1 -a 3
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [1d50:60e6 TARGET DEVICE *************************]
Sent power off request
New status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
$ sudo uhubctl
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0100 power
  Port 3: 0103 power enable connect [046d:c539 Logitech USB Receiver]
  Port 4: 0103 power enable connect [04d9:0296 USB-HID Keyboard]
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0489:e0a2 Foxconn / Hon Hai
Bus 001 Device 003: ID 0bda:58f4 Realtek Semiconductor Corp. Integrated_Webcam_HD
Bus 001 Device 007: ID 27c6:5385 Shenzhen Goodix Technology Co.,Ltd. Fingerprint Reader
Bus 001 Device 008: ID 04d9:0296 Holtek Semiconductor, Inc.
Bus 001 Device 006: ID 046d:c539 Logitech, Inc.
Bus 001 Device 010: ID 1d50:60e6 TARGET DEVICE *********************************
Bus 001 Device 002: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
$ device info
Traceback (most recent call last):
  ...
usb.core.USBError: [Errno 5] Input/Output Error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  ...
usb.core.USBError: [Errno 5] Input/Output Error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  ...
usb.core.USBError: [Errno 5] Input/Output Error

$ sudo uhubctl -p 1 -a 3
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
Sent power on request
New status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0101 power connect [1d50:60e6]
$ sudo uhubctl
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [1d50:60e6 TARGET DEVICE *************************]
  Port 2: 0100 power
  Port 3: 0103 power enable connect [046d:c539 Logitech USB Receiver]
  Port 4: 0103 power enable connect [04d9:0296 USB-HID Keyboard]
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0489:e0a2 Foxconn / Hon Hai
Bus 001 Device 003: ID 0bda:58f4 Realtek Semiconductor Corp. Integrated_Webcam_HD
Bus 001 Device 007: ID 27c6:5385 Shenzhen Goodix Technology Co.,Ltd. Fingerprint Reader
Bus 001 Device 008: ID 04d9:0296 Holtek Semiconductor, Inc.
Bus 001 Device 006: ID 046d:c539 Logitech, Inc.
Bus 001 Device 011: ID 1d50:60e6 TARGET DEVICE ***********************
Bus 001 Device 002: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
$ device-info
Found TARGET DEVICE *************************
Firmware info...
```
Is this intended behavior? It's important for me to be able to actually control the on/off state of the device for resets from the command line.

mvp, 2021-03-29:
Which version of uhubctl you are using? I recommend 2.4.0, or at least 2.3.0.
You have connected this USB3 hub as USB2 - this maybe a problem.
Can you make sure you are using native USB3 connection?

mvp, 2021-03-29:
For USB3 hubs it is very important to turn off both USB2 and USB3 virtual hubs inside to actually cut power off.
Many USB3 hubs are doing it correctly when connected as USB2 only.
However, some (and perhaps this hub too!) only work when connected as native USB3. In this case, uhubctl should detect both USB2 and USB3 virtual hubs and turn power off on both, killing VBUS.

If that does not help and hub does not turn off VBUS, it may mean that HU3641V1 was added to the list improperly. However, we usually require submitters to perform extensive checks before devices are added to compatibility list.

GravesJake, 2021-03-30:
Thanks for the response. I'm using `version: 2.4.0-3-ge5ae7b99`. The laptop I'm using this on only has USB-C ports, so I'm not sure what my best option is here. I have two adapters, one is USB3 and came with the laptop from Dell, the other is USB2 and came with a an android phone. 

When I try using the USB3 adapter, the `sudo uhubctl` command takes forever to run and ends up showing `No compatible devices connected` with `dmesg` showing things like:
```
[10369.271690] hid-generic 0003:04D9:0296.0028: input,hidraw7: USB HID v1.11 Mouse [USB-HID Keyboard] on usb-0000:39:00.0-2.3/input2
[10374.337262] xhci_hcd 0000:39:00.0: Timeout while waiting for setup device command
[10379.713251] xhci_hcd 0000:39:00.0: Timeout while waiting for setup device command
[10379.921204] usb 4-2: device not accepting address 5, error -62
[10379.929290] usb usb4-port2: unable to enumerate USB device
[10385.345266] xhci_hcd 0000:39:00.0: Timeout while waiting for setup device command
[10390.721261] xhci_hcd 0000:39:00.0: Timeout while waiting for setup device command
[10390.929133] usb 4-2: device not accepting address 6, error -62
[10396.353011] xhci_hcd 0000:39:00.0: Timeout while waiting for setup device command
```

However, it was as shown in my origincal post when using the USB2 adapter.

GravesJake, 2021-03-30:
Update: I plugged my HU3641V1 hub directly into my Kensington SD4600P USB-C dock (powered via external power supply), which is connected directly to one of my laptops two (lightning bolt compatible?) USB-C ports.

```
$ sudo uhubctl
Current status for hub 3-2.1.4 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [1d50:60e6 TARGET DEVICE ***********************]
  Port 2: 0103 power enable connect [046d:c539 Logitech USB Receiver]
  Port 3: 0103 power enable connect [04d9:0296 USB-HID Keyboard]
  Port 4: 0000 off
Current status for hub 3-2.1 [047d:8023 Kensington K38231_02 A1630A000209, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [04b4:3610 Kensington K38231_03 00A1630A000209]
  Port 2: 0103 power enable connect [08bb:2912 Burr-Brown from TI USB audio CODEC]
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
Current status for hub 3-2 [047d:8021 Kensington K38231_01 A1630A000209, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [047d:8023 Kensington K38231_02 A1630A000209, USB 2.10, 4 ports, ppps]
  Port 2: 0100 power
  Port 3: 0103 power enable connect [04b4:5217 Cypress Semiconductor Billboard Device 0001]
  Port 4: 0100 power
```
So I then tried toggling power to my target device
`sudo uhubctl -l 3-2.1.4 -p 1 -a 3` but I get the same result as before: I/O errors but target device remaining powered on.

mvp, 2021-03-31:
Thanks for checking this, but my original point still stands - please check if it works when HU3641V1 is connected as USB3. Your tests so far show that hub is visible only as USB2 device. When connected as USB3 device, it should show 2 devices (USB2+USB3), e.g. something similar to this:
```
Current status for hub 4-3 [2109:2811 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 3-3 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```
My current theory is that USB3 chip remains powered on when you connect as USB2 only and turn off USB2 ports only (this is a bug in itself, other hubs are typically handling this correctly).

GravesJake, 2021-03-31:
Oh gotcha, thanks for the extra clarity. Is there a way to ensure it's running as USB3 other than connecting directly to a port on the system? It's strange to me that it shows up as USB2 even though it's connected as `Device -> USB3 Hub -> USB3 port on dock -> USBC (lightning) laptop port`.

mvp, 2021-03-31:
It means something in your connection path is not capable of supporting USB3 speeds - it could be dock or cables.

mvp, 2021-03-31:
If you have access to any other computer with USB3 (Linux or Mac), easiest way to check is connect HU3641V1 to it directly (so you can simplify your connection path), make sure it is connected as USB3 and try controlling power in it. This will give you definitive answer if HU3641V1 has a chance to work.

I think I should add extra column to supported device table about device support confidence level. E.g. hubs that I checked myself would get checkmark, but it will be only 2 or 3 that I personally own :-)

GravesJake, 2021-03-31:
Thank you for all the help! I don't have access to another system right now but my target device doesn't contain a micro USB3 port so I think ultimately I'll have to look for a way to get it to work with USB2 right? Or is it possible that this hub connected to native USB3 will be able to cut power from my device connected as USB2?

GravesJake, 2021-03-31:
Thank you so much!

mvp, 2021-04-02:
If you could test HU3641V1 while connected to native USB3, that would be very helpful.
If it does not work in USB3, I would remove it from supported table.
If it works in USB3, I would add a note that it only works in USB3.
Thanks!

mvp, 2021-04-09:
Thanks for getting back to this @GravesJake!

No, hub should be completely reset after you physically disconnect it from host and reconnect again.

Can you please show output of commands you ran (perhaps repeat it again)?
Specifically, I am interested if it was able to properly identify and try to change state for both USB2 and USB3 components.
Also, if you see some weird stuff in `dmesg`, please post it too. If output is too long, might be wise to use gist.github.com. Feel free to censor end device names from your output, just please show everything else.
Thanks!

mvp, 2021-04-14:
Closing this by inactivity from submitter. @GravesJake please reopen if you have any update. Thanks!

closed 2021-04-14 by mvp

GravesJake, 2021-04-22:
Sorry for the delay, the hang up issue I was experiencing isn't occurring at the moment so I was able to do some testing:
```
person@gsgdemo4:~$ sudo uhubctl
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [1d50:60e6 Great Scott Gadgets GreatFET 000000000000000057cc67e630125257]
person@gsgdemo4:~$ sudo uhubctl -p 4 -a 0
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0503 power highspeed enable connect [1d50:60e6 Great Scott Gadgets GreatFET 000000000000000057cc67e630125257]
Sent power off request
New status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
person@gsgdemo4:~$ sudo uhubctl
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0000 off
person@gsgdemo4:~$ sudo uhubctl -p 4 -a 0
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
Sent power off request
New status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
person@gsgdemo4:~$ sudo uhubctl
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0000 off
person@gsgdemo4:~$ sudo uhubctl -l 2-1 -p 4 -a 0
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
Sent power off request
New status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
person@gsgdemo4:~$ sudo uhubctl
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0000 off
```

It doesn't seem to be properly turning off the USB3.0 2-1 port along with USB2.0 1-1 or even when trying to limit it using `-l 2-1`.

I'll update with dmesg output if I run into the previous hang up issue again.

mvp, 2021-04-23:
@GravesJake, it sounds like hub does not want to turn off your target device connected to port 4 of hub `1-1` (or `2-1` which is USB3 instance of the same physical port).

There is very high possibility that this is linux kernel issue as described in FAQ https://github.com/mvp/uhubctl#power-comes-back-on-after-few-seconds-on-linux:
> Some device drivers in kernel are surprised by USB device being turned off and automatically try to power it back on.
> You can use option -r N where N is some number from 10 to 1000 to fix this - uhubctl will try to turn power off many times in quick succession, and it should suppress that. This may be eventually fixed in kernel, see more discussion [here](https://bit.ly/2JzczjZ).

Other than that, I recommend testing if power can be turned off when target device is physically disconnected. If that works, it will be another confirmation of this issue. `dmesg -T` should provide even more clues - e.g. if you see kernel complaining about device disconnected and attempting to revive it.

Note that this is Linux only problem, and should not be present on other OS like Mac or BSD.

GravesJake, 2021-04-23:
I unfortunately started having the hang up issue again:
```
[Fri Apr 23 14:58:48 2021] usb 1-1: new high-speed USB device number 4 using xhci_hcd
[Fri Apr 23 14:58:48 2021] usb 1-1: New USB device found, idVendor=2109, idProduct=2811, bcdDevice=90.70
[Fri Apr 23 14:58:48 2021] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Fri Apr 23 14:58:48 2021] usb 1-1: Product: USB2.0 Hub             
[Fri Apr 23 14:58:48 2021] usb 1-1: Manufacturer: VIA Labs, Inc.         
[Fri Apr 23 14:58:48 2021] hub 1-1:1.0: USB hub found
[Fri Apr 23 14:58:48 2021] hub 1-1:1.0: 4 ports detected
[Fri Apr 23 14:58:53 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:58:59 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:58:59 2021] usb 2-1: device not accepting address 2, error -62
[Fri Apr 23 14:58:59 2021] usb 1-1.4: new high-speed USB device number 5 using xhci_hcd
[Fri Apr 23 14:58:59 2021] usb 1-1.4: New USB device found, idVendor=1d50, idProduct=60e6, bcdDevice= 1.00
[Fri Apr 23 14:58:59 2021] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Apr 23 14:58:59 2021] usb 1-1.4: Product: <DEVICE>
[Fri Apr 23 14:58:59 2021] usb 1-1.4: Manufacturer: <COMPANY>
[Fri Apr 23 14:58:59 2021] usb 1-1.4: SerialNumber: 000000000000000057cc67e630125257
[Fri Apr 23 14:59:04 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:59:10 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:59:10 2021] usb 2-1: device not accepting address 3, error -62
[Fri Apr 23 14:59:10 2021] usb usb2-port1: attempt power cycle
[Fri Apr 23 14:59:16 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:59:21 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:59:22 2021] usb 2-1: device not accepting address 4, error -62
[Fri Apr 23 14:59:27 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:59:32 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:59:33 2021] usb 2-1: device not accepting address 5, error -62
[Fri Apr 23 14:59:33 2021] usb usb2-port1: unable to enumerate USB device
[Fri Apr 23 14:59:38 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:59:43 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:59:44 2021] usb 2-1: device not accepting address 6, error -62
[Fri Apr 23 14:59:49 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:59:54 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 14:59:55 2021] usb 2-1: device not accepting address 7, error -62
[Fri Apr 23 14:59:55 2021] usb usb2-port1: attempt power cycle
[Fri Apr 23 15:00:01 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:00:06 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:00:06 2021] usb 2-1: device not accepting address 8, error -62
[Fri Apr 23 15:00:12 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:00:17 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:00:17 2021] usb 2-1: device not accepting address 9, error -62
[Fri Apr 23 15:00:17 2021] usb usb2-port1: unable to enumerate USB device
[Fri Apr 23 15:00:23 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:00:28 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:00:28 2021] usb 2-1: device not accepting address 10, error -62
[Fri Apr 23 15:00:34 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:00:39 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:00:39 2021] usb 2-1: device not accepting address 11, error -62
[Fri Apr 23 15:00:40 2021] usb usb2-port1: attempt power cycle
[Fri Apr 23 15:00:46 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:00:51 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:00:51 2021] usb 2-1: device not accepting address 12, error -62
[Fri Apr 23 15:00:57 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:01:02 2021] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Apr 23 15:01:02 2021] usb 2-1: device not accepting address 13, error -62
[Fri Apr 23 15:01:02 2021] usb usb2-port1: unable to enumerate USB device
[Fri Apr 23 15:01:23 2021] usb 1-1.4: USB disconnect, device number 5
[Fri Apr 23 15:01:28 2021] usb 1-1: USB disconnect, device number 4
[Fri Apr 23 15:01:33 2021] usb 1-1: new high-speed USB device number 6 using xhci_hcd
[Fri Apr 23 15:01:33 2021] usb 1-1: New USB device found, idVendor=2109, idProduct=2811, bcdDevice=90.70
[Fri Apr 23 15:01:33 2021] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Fri Apr 23 15:01:33 2021] usb 1-1: Product: USB2.0 Hub             
[Fri Apr 23 15:01:33 2021] usb 1-1: Manufacturer: VIA Labs, Inc.         
[Fri Apr 23 15:01:33 2021] hub 1-1:1.0: USB hub found
[Fri Apr 23 15:01:33 2021] hub 1-1:1.0: 4 ports detected
[... cut ...]
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0503 power highspeed enable connect [1d50:60e6 <COMPANY> <DEVICE> 000000000000000057cc67e630125257]
Sent power off request
New status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
person@computer:~$ sudo uhubctl
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0000 off
```
Which doesn't do me any good.

GravesJake, 2021-04-23:
I did find an interesting line in dmesg:
```
[Fri Apr 23 15:17:12 2021] hub 1-0:1.0: USB hub found
[Fri Apr 23 15:17:12 2021] hub 1-0:1.0: 12 ports detected
[Fri Apr 23 15:17:12 2021] xhci_hcd 0000:00:14.0: xHCI Host Controller
[Fri Apr 23 15:17:12 2021] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 2
[Fri Apr 23 15:17:12 2021] xhci_hcd 0000:00:14.0: Host supports USB 3.0 SuperSpeed
[Fri Apr 23 15:17:12 2021] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.08
[Fri Apr 23 15:17:12 2021] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[Fri Apr 23 15:17:12 2021] usb usb2: Product: xHCI Host Controller
[Fri Apr 23 15:17:12 2021] usb usb2: Manufacturer: Linux 5.8.0-50-generic xhci-hcd
[Fri Apr 23 15:17:12 2021] usb usb2: SerialNumber: 0000:00:14.0
[Fri Apr 23 15:17:12 2021] hub 2-0:1.0: USB hub found
[Fri Apr 23 15:17:12 2021] hub 2-0:1.0: 6 ports detected
[Fri Apr 23 15:17:12 2021] usb: port power management may be unreliable
```
"port power management may be unreliable"?

mvp, 2021-04-23:
I have never seen this before.

Continued messages like this:
> [Fri Apr 23 15:00:39 2021] usb 2-1: device not accepting address 11, error -62

are most likely some issues with your end device misbehaving.

As for `port power management may be unreliable`, it must be coming from this kernel code: 
https://github.com/torvalds/linux/blob/3db1a3fa98808aa90f95ec3e0fa2fc7abf28f5c9/drivers/usb/core/port.c#L400-L401.
Especially concerning is this line:
> usb_port_block_power_off = 1;

I really don't know what to say here. It could be your end device misbehaving, it could be faulty USB hub.

At the end of the day, all that uhubctl does is to send this control message directly to the hub to change power state: 
https://github.com/mvp/uhubctl/blob/7bafc7ffb388e0fc51ccb219d564ea8ec53b4770/uhubctl.c#L1074-L1078

So this control message is routed down to kernel USB driver.

Could you try connecting your hub to any other OS host, e.g. Mac or FreeBSD? I expect it to work correctly there.

GravesJake, 2021-04-26:
I've seen the `device not accepting address x` on my other ubuntu laptop as well, been a huge pain for me. But yes I'll install a different OS and try my luck there. Thanks!