mvp/uhubctl #331: Linksys USB2HUB4

ChaoticEnigma, 2021-04-02:
I know it's already listed, but I wanted to add some details I might have about the Linksys USB2HUB4.

    Current status for hub 3-2.1 [0409:0058 NEC Corporation USB2.0 Hub Controller, USB 2.00, 4 ports, ppps]
      Port 1: 0100 power
      Port 2: 0100 power
      Port 3: 0100 power
      Port 4: 0100 power

I am able to able to disable the VBUS on each port with `uhubctl -L 3-2.1 -p 1 -a toggle`, tested with a USB voltage/current monitor (not a USB device, only looks at VBUS).

Also interesting, if I overload the VBUS on one of the ports (e.g. with a 1A load), the hub temporarily disables the port without affecting the others, and the port is reported as "off". As below, where port 3 is overloaded:

    Current status for hub 3-2.1 [0409:0058 NEC Corporation USB2.0 Hub Controller, USB 2.00, 4 ports, ppps]
      Port 1: 0103 power enable connect [1050:0120 Yubico Security Key by Yubico]
      Port 2: 0100 power
      Port 3: 0000 off
      Port 4: 0100 power

And it seems to return to normal when the overload is removed.

mvp, 2021-04-02:
Thanks, this is actually expected behavior for properly implemented USB hub.
Hub should sense OC (overcurrent) condition and disable port for misbehaving device.
Nevertheless, I will add a note in hardware table linking to this - thank you!

closed 2021-04-02 by mvp

ChaoticEnigma, 2021-04-02:
Yes, the overcurrent behavior was expected. But I was a little surprised that the overcurrent disabled state was visible to the host.
You could also add the VID and PID to the hardware table.

mvp, 2021-04-02:
Thanks, I will add vid:pid when I need to update table next time.

uhubctl is querying hub directly with USB control messages to get or set power state - thus is why we can see current state properly (provided that USB chip implements it correctly).

commit 7bafc7ff (2021-04-02): Add a note for Linksys USB2HUB4

Closes #331.