mvp/uhubctl #336: Generic linux USB hub 1d6b:0003 with no power switching does switching

splatch, 2021-04-08:
Thank you for awesome project. I hope to solve with your project help some issues I have with serial USB dongle. Actual device is Aaeon Boxer mini pc based on Intel components. I couldn't find more details on hub controller.

As part of verification trail I did run 2.4.0 version:
```
$ uhubctl -v 
2.4.0
$ uhubctl   
No compatible devices detected!
Run with -h to get usage info.
$ uhubctl -f
Current status for hub 2 [1d6b:0003 Linux 4.19.80-intel-pk-standard xhci-hcd xHCI Host Controller 0000:00:15.0, USB 3.00, 7 ports, nops]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
  Port 5: 02a0 power 5gbps Rx.Detect
  Port 6: 02a0 power 5gbps Rx.Detect
  Port 7: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 4.19.80-intel-pk-standard xhci-hcd xHCI Host Controller 0000:00:15.0, USB 2.00, 8 ports, nops]
  Port 1: 0103 power enable connect [0403:6001 FTDI USB-RS485 Isolated Converter]
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [05ac:12a8 Apple Inc. iPhone]
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0503 power highspeed enable connect [05c6:9215 Quectel Quectel LTE Module]
  Port 8: 0100 power
```

I have a custom Linux system image which limits amount of available kernel modules/drivers so it might be wrong marker.  Interestingly vendor id `1d6b:0003` does match supported one yet it is not recognized as compatible for power switching. After doing test it turns it actually works.

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            4.19
  iManufacturer           3 Linux 4.19.80-intel-pk-standard xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:00:15.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
[... 162 lines cut ...]
  bLength              11
  bDescriptorType      41
  nNbrPorts             8
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x80 0x01
  PortPwrCtrlMask    0xff 0xff
 Hub Port Status:
   Port 1: 0000.0103 power enable connect
   Port 2: 0000.0100 power
   Port 3: 0000.0503 highspeed power enable connect
   Port 4: 0000.0100 power
   Port 5: 0000.0100 power
   Port 6: 0000.0100 power
   Port 7: 0000.0503 highspeed power enable connect
   Port 8: 0000.0100 power
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

mvp, 2021-04-08:
vendor id 1d6b:0003 is generic id used by linux kernel for root USB3 hubs. Note that linux support for root USB3 hubs is strange in general, for example they don't advertise bos container id, which makes it more difficult to find USB2/USB3 hub matching (absolutely required to control power properly).

 In general, we cannot rely on vid:pid to recognize supported devices, as many devices use the same USB chip and don't change its vid:pid, yet may use different circuitry.

Another point here us that uhubctl showing power is off is not good indicator it is actually off. The only reliable way to tell is to measure actual voltage, or check if phone stops charging, usb light stops shining or usb fan stops spinning.

splatch, 2021-04-09:
I did test with phone charger and power goes off. In my first attempt I used serial adapter with a LED to see if it goes dark. Not sure why, but LED on serial adapter is always on, even if power goes off.

I am quite happy with outcomes, so feel free to close issue. Thank you very much for awesome work!

mvp, 2021-04-09:
Hmm. Maybe there is still tiny current present that is enough to light led for serial adapter?
Anyway, thanks for testing!

closed 2021-04-09 by mvp