mvp/uhubctl #343: Working around there being very few hubs with PPPS wired up.

fifteenhex, 2021-05-01:
Hi,

So I spent a week or more looking for an off the shelf hub that did PPPS.
I couldn't find anything here in Japan. But it seems some ~$20 USB3 hubs can be made to work with uhubctl with some soldering.

I hacked up a RSH-516-4 here: https://github.com/fifteenhex/diyppps

Cheers,

Daniel

mvp, 2021-05-01:
Thanks for your report, this sounds cool.

However, I am not sure how many people are willing or capable of going that far to solder things - I am certainly not one of them.

I am lucky owner of another RSHTECH hub RSH-A16 (16 port, nominal cost is about $70), which RSHTECH sent me last year so I could fix uhubctl to work with any USB 3 hubs correctly. It works fine without any soldering. Perhaps you can find one in Japan?

Regarding your report, I will add a section to readme about your hack to convert some hubs so they become PPPS compliant. Thanks!

fifteenhex, 2021-05-01:
Hi,

Does the RSH-A16 support turning off any of the 16 ports? From the note I read it looked like it could only control sets of 4 ports.
I went for the 4 port version in the hope that they left the PPPS support with the ability to control each port.

Thanks,

Daniel

mvp, 2021-05-01:
I tested it, and does seem to work individually on all 16 ports. I tested by connecting Android phone and confirming that it stops charging once port is powered down.

fifteenhex, 2021-05-01:
Interesting. I might have to get one of those too if I can find one. :)
I wonder why they didn't wire it up in the smaller ones :(

Thanks.

mvp, 2021-05-01:
This is quote from RSHTECH support email where I asked them why RSH-A37 doesn't work when RSH-A16 does. They answered:
> On Fri, Dec 11, 2020, 3:32 AM RSHTECH Team <support@rshtech.com> wrote:
Regarding the RSH-A37 USB Hub, the customer bought is RSH-A37B USB hub with four data ports and three fast charging ports. Our hub uses the same data port chip, but the fast charge port chip is different.

commit 23c15ffc (2021-05-02): Add diyppps to notable projects

Closes #343.
closed 2021-05-02 by mvp