mvp/uhubctl #344: 4-Port USB-3 ORICO M3H4-V1-SV

CyberThunk, 2021-05-07:
4-Port USB-3 ORICO M3H4-V1-SV
 All 4 ports powered on and off with uhubctl on MacBook 12" Retina early 2015 running macOS 11.3.1.

mvp, 2021-05-07:
Thanks for your report! Can you please check for vbus off support? Easiest way is to plug a phone and check that it stops charging once you power port down.
Too many hubs out there would cut off data connection, but still keep voltage present - we don't add such hubs. Thank you!

CyberThunk, 2021-05-07:
Only stops data transmission   Power still live.

CyberThunk, 2021-05-07:
A bit more info. This device does power off the port.  On my phone the charging appears to continue when the port is powered off but if  I unplug it and then plug it back into the powered off port it will not charge.  

I have a Plugable 7-port Hub  USB3-HUB7C that exhibits the exact same behavior as mentioned above on the ORICO hub. 

The Plugable is one of the models listed on your page.

mvp, 2021-05-08:
We try to list only hubs which can actually turn power off reliably.

Regarding Plugable USB3-HUB7C, according to original report #171, true power control is only working on port 4 (perhaps it is charging port?).

mvp, 2021-05-08:
Actually, there is another report #322 which states that Plugable USB3-HUB7C supports true power control only on 2 charging ports.

mvp, 2021-05-08:
Anyway, I am closing this, as Orico M3H4-V1-SV is not supporting vbus control.

closed 2021-05-08 by mvp