mvp/uhubctl #346: Delock 62537 works

FauthD, 2021-05-15:
Hi, 
this is a USB3 hub with 3 USB A and one USB C downstream ports.
Upstream has a short cable with USB C connector.

FauthD, 2021-05-15:
Sorry, sent to fast.
here is the link to dealer:
https://www.delock.de/produkt/62537/merkmale.html?setLanguage=en

Many regards, Dieter Fauth

mvp, 2021-05-16:
Can you please check if it supports turning VBUS off?

FauthD, 2021-05-17:
Hi, 
sure it does. Otherwise I would not have written this report.
Checked with "charge doctors" and an WD USB drive. Its LED goes off as well.
A few other hubs I checked did not turn off vbus.
Also asked on support site of delock whether they know other models. Got a response, but he did not even understand my question. Will try to clarify.
Dieter

mvp, 2021-05-18:
Thank you! Added to the list in f20007e.

commit f20007e2 (2021-05-18): Add Delock 62537 as compatible device

Closes #346.
closed 2021-05-18 by mvp