mvp/uhubctl #349: Getting error when running "make" command on Fedora

skanipakala, 2021-06-01:
I have downloaded the source code and installed all the dependencies you mentioned, but getting errors when trying to compile using the 'make' command. I have also tried with sudo, still failed. 

Here is a link to the screenshot:
https://lh3.googleusercontent.com/--8NVIv6Btl8/YLWjK-rDrXI/AAAAAAAAwJM/S81WUFLRdUM3PBDJclclFC-r7la8wvwCQCJEEGAsYHg/s0/2021-05-31.jpg?authuser=0

mvp, 2021-06-01:
Make sure you have installed libusb version 1.0.12 or later, 1.0.22 or later is recommended. On Fedora/Redhat, you usually need to execute:
```
sudo yum install libusb1-devel
```
Check which version you actually have installed using
```
rpm -qa libusb*
```

skanipakala, 2021-06-01:
I have run those commands already, all packages were up to date. Here is the screenshot, still getting ERROR. Thanks for helping me.

https://lh3.googleusercontent.com/-8ufbgvCrVpY/YLWny_6H8TI/AAAAAAAAwJg/VBIhKRZSW-47BtqeqZik9hRMLs0p8mp3ACJEEGAsYHg/s0/2021-05-31.jpg?authuser=0

mvp, 2021-06-01:
Your screenshot shows Fedora 14 - which is too old, it has libusb 1.0.8, and uhubctl requires at least libusb 1.0.12, 1.0.22+ recommended.

I do not have access to Fedora system right now, but on Centos 7 I get:
```
$ sudo yum install libusb1-devel
Package libusbx-devel-1.0.21-1.el7.x86_64 already installed and latest version
```
You might be able to get uhubctl to build even on your old system if you compile libusb from sources and use correct compilation options. But, I recommend to simply upgrade your [10 year old OS](https://en.wikipedia.org/wiki/Fedora_version_history#Fedora_14) to something newer.

closed 2021-06-01 by mvp