mvp/uhubctl #352: UNITEK Y-3098ABK

264768502, 2021-06-24:
It support well with uhubctl, VBUS also work
product introduce: https://www.unitek-products.com/products/usb3-0-4-port-hub

mvp, 2021-06-24:
Thanks for reporting!

Can you please provide output for `sudo uhubctl` with this hub?

264768502, 2021-07-07:
```
Current status for hub 2-4 [0bda:0411 Generic 4-Port USB 3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-4 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

closed 2021-07-07 by mvp

lzs2000131, 2021-07-29:
Your title is Y-3098 but you give a product link of Y-3089, are you sure they are the same?
I bought a Y-3098ABK, it's not working correctly. Maybe it's because my computer doesn't support USB3.0?

mvp, 2021-07-29:
Huh, good catch @lzs2000131 . It was reported by @264768502 , I do not have access to this hub.
If you are sure it does not work, I would rather remove it from supported list.
However, I recommend trying it on computer with USB3 first, it should work on most Linux and Mac computers.

264768502, 2021-07-29:
Sorry, my local model is UNITEK Y-3098ABK. I just try Debian 10. It definitely work.
Actually, I tried USB2.0 port.

but we don't know whether chip is aligned in China or out.
I get hub by https://item.jd.com/2138009.html, not vendor formal site.

lzs2000131, 2021-07-29:
I bought from jd too, the same link (233333)
I'm a mac user, I found another Macbook air support USB3 and turned off all ports, but my lights are still on!
[image]
Maybe it's working on linux, not on mac

mvp, 2021-08-03:
Thanks for reporting @lzs2000131. If it doesn't work on Mac, it will not work period - it means VBUS off is not really supported. I will remove this hub from supported list.

commit 5ba647ee (2021-07-07): Add Unitek Y-3098ABK as compatible device

Closes #352.