mvp/uhubctl #353: Rapsberry pi 4b with usb fan issues

evanlaf, 2021-06-25:
Hello, 

I am running the command `uhubctl -l 2 -a 0` on my raspberry pi 4B and am running into an issue. I currently have a usb mouse/keyboard plugged into a usb 2 slot, and a usb fan plugged into the other usb 2 slot. My longterm goal is to trigger the fan on/off using the usb power, but it doesn't seem to turn off. For example, when I run the command `uhubctl -l 2 -a 0`, it shows that there is no power to all 4 ports and the keyboard/mouse stop working. Which is expected. But, the usb fan still keeps on going. Does anyone have an idea on how to fix this? 

Thanks.

mvp, 2021-06-25:
Make sure you have updated RPi4 firmware as stated in readme. Also, make sure you use latest uhubctl version (2.4.0), anything older than 2.3.0 will not work correctly with RPi4.

evanlaf, 2021-06-25:
> Make sure you have updated RPi4 firmware as stated in readme. Also, make sure you use latest uhubctl version (2.4.0), anything older than 2.3.0 will not work correctly with RPi4.

@mvp Sorry I should have included version numbers in the original post. I currently have version `00138a1` and the bootloader is up to date. Original issue was I soft installed uhubctl (`sudo apt install uhubctl`) and that gave me version 2.0.0 instead of 2.4.0. That's fixed now, so the different issue I'm now running into is --- I am unable to turn off individual ports. When I run any command similar to `uhubctl -l 1-1 -p <PORT> -a 0` (where PORT is either 1, 2, 3, or 4), it sends the power off signal but no ports actually turn off. Only by calling `uhubctl -l 1-1 -a 0` does every single port actually turn off. 

I realize this is probably a limitation in the RPi4 (as stated in the section _Raspberry Pi turns power off on all ports, not just the one I specified_), but wanted to confirm or ask if there is a workaround rather than close the issue immediately.

mvp, 2021-06-25:
You cannot control ports individually on Rpi, this is hardware limitation, sorry.

closed 2021-06-25 by mvp