mvp/uhubctl #356: Confirming Hardware compatability - Anker USB 3 hub

ronenrozner, 2021-07-13:
I want to add to the confirmed hardware working with usubctl an Anker USB 3.0 4-Port Portable Aluminum Hub. 

lsusb output: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub

mvp, 2021-07-13:
We have had previous reports with other Anker hubs, and not all of them actually worked, e.g. this one #169.

Can you please check if your hub supports VBUS off as [described in the FAQ](https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl)?

ronenrozner, 2021-07-13:
> We have had previous reports with other Anker hubs, and not all of them actually worked, e.g. this one #169.
> 
> Can you please check if your hub supports VBUS off as [described in the FAQ](https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl)?

You are right. The issue is that the power comes back after a few seconds. I guess it's a no-go for this hub.

mvp, 2021-07-13:
Not necessarily. If power (voltage) does go down even momentarily, it is a good sign it might just work.
If it switches back on after few seconds, this could be different issue and it not a hub fault https://github.com/mvp/uhubctl#power-comes-back-on-after-few-seconds-on-linux.
Try adding higher `-r N` option to suppress this.

mvp, 2021-07-21:
Closing for inactivity from @ronenrozner. Please reopen if this command works for you:
```
sudo uhubctl -l {hub} -p {port} -a 0 -r 1000
```

closed 2021-07-21 by mvp