mvp/uhubctl #358: not working when called from script

erikvoznak, 2021-07-21:
Hello, I am using uhubctl to reset usb touch interface connected via USB to my rpi4. when uhubctl is running from script, from the logs it seems that it is doing same thing when running from putty session.  however, when running from script it totally does nothing...

Can you please help me with this really strange issue?

how it is executed from script:

    /home/pi/rpi-hdmi.sh on
    sleep 360
    su - root -c "uhubctl -l 1-1 -p 4 -a 2"
    echo "$timestamp starting display because erik is home" >> /var/log/mirrorstatus.log

log from script:

Powering on HDMI with preferred settings
Current status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 4: 0080 off
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0103 power enable connect [0408:3001 QUANTA OpticalTouchScreen]
Sent power off request
New status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
Current status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 4: 0080 off
Sent power on request
New status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0101 power connect [0408:3001]
Sent power on request
New status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0103 power enable connect [0408:3001]

when trigerred from putty session:

`root@mirror:~ # uhubctl -l 1-1 -p 4 -a 2
Current status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 4: 0080 off
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0103 power enable connect [0408:3001 QUANTA OpticalTouchScreen]
Sent power off request
New status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
Current status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 4: 0080 off
Sent power on request
New status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0101 power connect [0408:3001]
Sent power on request
New status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0103 power enable connect [0408:3001]
root@mirror:~ #`

mvp, 2021-07-21:
Rpi4B doesn't support power off only on one port. You have to reset all ports. You need to make sure you use uhubctl 2.4.0, also read FAQ regarding USB firmware update and use this command:
```
uhubctl -l 1-1 -a 2

closed 2021-07-21 by mvp

erikvoznak, 2021-07-22:
your response does not answer my problem.
when I execute uhubctl -l 1-1 -p 4 -a 2 from putty session it works as it should. 
when it is executed from script it does NOT work. 
thats my problem.

mvp, 2021-07-22:
On raspberry pi 4, `uhubctl -l 1-1 -p 4 -a 2` should not work because to toggle power, you have to toggle all ports `uhubctl -l 1-1 -a 2`.

When you talk about it not working from script (regardless of port 4 vs all ports), make sure you specify full path to uhubctl, e.g. /usr/sbin/uhubctl - your script may not have the same PATH as your terminal session does. Also, I recommend to configure USB permissions as per FAQ, because it will make it work without sudo, which could be another reason why it doesn't work from script.

erikvoznak, 2021-07-23:
strange, I changed in script
from:
su - root -c "uhubctl -l 1-1 -p 4 -a 2"
to:
su - root -c "/usr/sbin/uhubctl -l 1-1 -p 4 -a 2"

and it finally works also in script. I compaired env variables and they are same in putty session as well as in 'su - root -c ""'
also output in log is same, no matter if uhubctl is called with fullpath or not, but it only works when called with fullpath.
nevermind, thx for help.