mvp/uhubctl #359: raspberry pi 4 usb power off not working continuously

cvcachagua, 2021-07-22:
I would try and power the usb off, and this would work, but for less than a few seconds.
I would try to update the power/authorized file and this would work until I used uhubctl and then the file would be overwritten, I guess by some process/kernel.

I solved this by using the kernel setting in the raspberry pi boot:
/boot/cmdline.txt adding coreusb.authorized_default=0 at the end of the commands there, then rebooted.
I was then able to use uhubctl after making the above modification.  thought someone may benefit.

mvp, 2021-07-22:
Did you try using higher -r option as described in FAQ https://github.com/mvp/uhubctl#power-comes-back-on-after-few-seconds-on-linux ?

cvcachagua, 2021-07-23:
I tried that, but it did not help.  Only the kernel flag for boot up.  Many of the users in our group do not have the issue, but for some reason, some do.  Possibly due to "apt upgrade".

cvcachagua, 2021-07-23:
Some users say that the github uhubctl works vs what my experience was using: apt install uhubctl.
BTW:
cat /proc/version 
Linux version 5.4.83-v7l+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1379 SMP Mon Dec 14 13:11:54 GMT 2020
The kernel boot parameters were found on: https://www.kernel.org/doc/html/v5.0/admin-guide/kernel-parameters.html

mvp, 2021-07-24:
Yes, you have to use uhubctl version 2.4.0 to make it work

mvp, 2021-07-28:
Closing for inactivity from submitter. Please feel free to reopen if you have anything to add.

At the minimum, make sure you are running uhubctl 2.4.0, compile it from source if necessary - follow instructions in readme. version installed by apt is very likely to be older.

closed 2021-07-28 by mvp