mvp/uhubctl #361: Anker PowerExpand 6-in-1 USB-C PD Ethernet Hub

scosu, 2021-08-31:
Just discovered, that this Anker USB-C Hub with an Ethernet port and power delivery can switch on and off all ports.

```
> lsusb
Bus 003 Device 014: ID 291a:8365 Anker Innovations Limited Anker USB-C Hub Device

> uhubctl
Current status for hub 3-4 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [067b:2303 Prolific Technology Inc. USB-Serial Controller]
  Port 5: 0503 power highspeed enable connect [291a:8365 Anker Innovations Limited Anker USB-C Hub Device 0000000000000001]
Current status for hub 2-3 [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.10, 4 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [0bda:8153 Realtek USB 10/100/1000LAN 000001]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
```

If you need further information, let me know. Thanks for the project :)

scosu, 2021-08-31:
Forgot the modelnumber as stated by amazon:
‎AK-A83650A1

mvp, 2021-09-01:
Thanks for your report! It's weird that this hub is advertising 5 ports for USB 2 part vs 4 ports for USB 3 part - that will disable automatic USB 2/3 duality handling. Can you please check if this hub supports turning off power? You will need to plug in phone and turn off same port number (1 to 4) on both USB 2 and USB 3 chips - phone should stop charging. If it doesn't, we cannot consider it supported device.

scosu, 2021-09-21:
@mvp Sorry for the delay.

I checked with the power-lane of a raspberry pi zero instead of a phone on the two USB-A ports. I could power off/on both ports (in particular these were ports 3 and 4).

mvp, 2021-09-29:
Thanks, added this hub to the list. Note that this hub will need to use 2 commands (one for USB2, another for USB3) to turn power off or on, e.g.

    uhubctl -e -l 3-4 -p 4 -a 0
    uhubctl -e -l 2-3 -p 4 -a 0

This is necessary because automatic USB2/USB3 pair detection will not work as this hub weirdly reports 4 ports for USB2 but 5 ports for USB3.

closed 2021-09-29 by mvp

scosu, 2022-01-27:
> @sco – May I ask, did this work on Mac, or just Linux?

I don't have a mac, so I only tested Linux.

mvp, 2022-01-27:
@gouldenstein , I have seen lots of hubs, and never seen one that works on just one OS. it if works, it works on any OS.

jdk, 2022-05-25:
Big disappointment. Just bought one, and although it shows the ports are off, my target USB board still has LEDS on and I can see and talk with it as well...

```
Current status for hub 2-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
  Port 2: 00a0 off
  Port 3: 00a0 off
  Port 4: 00a0 off
Current status for hub 2-1 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 5 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
  Port 5: 0000 off
```

scosu, 2022-05-31:
Hi there, not sure what's different. I can test this hub in maybe 3 or 4
weeks again.

darkez, 2022-08-08:
> and although it shows the ports are off, my target USB board still has LEDS on and I can see and talk with it as well...

i have the same model, macos 10.15 and i have the same behaviour. Have you solved?

enwony, 2023-04-20:
Same behaviour for same model for me.
Also I discovered that specific usb 1.1 device not even detected.

scosu, 2023-04-23:
Hi, my hub broke and I am not going to replace it so can't test it anymore.
If it doesn't work for multiple people maybe it is better to remove it again?

ChrisVilches, 2023-09-17:
This hub is not supported. I turned off all items in the list, and my USB lamp is still ON.

mvp, 2023-09-23:
Due to so many negative reports I have removed this hub from supported list.

commit 75b1f668 (2021-09-29): Add Anker AK-A83650A1 as compatible device

Closes #361.